clnhub/rtl8192eu-linux #4: Question

arj03, 2019-05-15:
Hi

I have a edimax EW-7612UAn wifi card and have been looking for a driver. I tried the device on a windows machine to see what kind of driver it would load and it loaded a rtl8192 driver so I assumed this driver would work for the device. Sadly it doesn't, so I went to the manufacturers page too see which drivers they had (available)[https://www.edimax.com/edimax/download/download/data/edimax/global/download/for_home/wireless_adapters/wireless_adapters_n300/ew-7612uan_v2] and there is a driver that seems to be for rtl 8188C / 8192C that with some minor [fixes](https://github.com/arj03/edimax-EW-7722UTN_V2/) works, but spits out tons of debug in dmesg. Strangely enough there is under the GPL section a RTL8192 driver that I can't seem to compile.

I was wondering if you could help me out figuring what the difference is between these different drivers.

Thanks! :)

kimocoder, 2019-05-15:
Jeg har også en driver til den, pakket med monitor mode å frame injection. Finner den i repoene mine.

Problemet er at alle Realtek drivere er meget like, jeg har flere drivere jeg maintainer, men de bruker dessverre eldre I/O handlere, de bruker nl80211 kombinert med cfg80211 istedenfor nyere mac80211. Så har du alle patchene som varierer per repp, ellers er de fleste RTL drivere (basen) meget, meget lik på tvers av driverne.

clnhub, 2019-05-16:
This driver only supports 8192EU. You can try [pvaret](https://github.com/pvaret/rtl8192cu-fixes)'s driver if your chipset is 8192CU [.](https://github.com/pvaret/rtl8192cu-fixes).

closed 2019-05-17 by arj03