clnhub/rtl8192eu-linux #13: binary package for rtl8192eu: 1.0 not found

DobleV55, 2019-12-10:
I want to install it, but I get this error when I make:

`sudo dkms install rtl8192eu/1.0`

```
Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' all KVER=5.0.0-37-generic........(bad exit status: 2)
ERROR (dkms apport): binary package for rtl8192eu: 1.0 not found
Error! Bad return status for module build on kernel: 5.0.0-37-generic (x86_64)
Consult /var/lib/dkms/rtl8192eu/1.0/build/make.log for more information.
```

clnhub, 2019-12-16:
If you are on 18.04 it should work now, otherwise could you post the log part with the error?

willianjusten, 2020-01-01:
I'm having exactly the same issue but on `5.4.7-050407-generic`.

clnhub, 2020-01-07:
Is this on 18.04?

Did you try the installer?

willianjusten, 2020-01-12:
Yeah, it was on 18.04, the only way to make it work was going back to kernel 5.3.0. It's working fine now and I don't need to use the latest Kernel, so I'm okay with it.

clnhub, 2020-01-13:
It should work though. I merged a specific pull request to enable kernel 5.4 support for 18.04 a while ago.

Did you completely remove the dkms driver with 

> sudo rmmod 8192eu
> sudo dkms remove rtl8192eu/1.0 --all

and re-install after installing kernel 5.4 (from the driver directory) with

> sudo dkms add .

before running

> sudo dkms install rtl8192eu/1.0

?

willianjusten, 2020-01-13:
Yeah, I did all the procedures to remove and install again but it was showing exactly the same error. I was using Ubuntu 18.04.3 with the latest Kernel...

clnhub, 2020-01-14:
Ok, I can repro on 18.04. This is related to using an older gcc version <7.5 which is not compatible with kernel 5.4 ('s asm_inline). 

The issue will be fixed once gcc 7.5 is released (already in proposed packages).

fdhcastillo, 2020-05-03:
i have gcc 7.5 and the same problem here

davehor, 2020-12-11:
Newest kernel 5.4.0-56-generic same problem and the dongle has stopped working altogether

clnhub, 2020-12-12:
Did you try an older 5.4 or the default 5.4.83 kernel?

davehor, 2020-12-12:
So my Comment got lost. 53 didn't work either had to go back to 52 kernel to get it to work again.

clnhub, 2020-12-12:
Log? Output of `cat /proc/version` on the kernel with issues?

Did you try the unmodified 5.4.83 kernel from the mainline directory, or a newer one? It compiles fine on 20.10.

Rahulgutal, 2021-08-18:
Could you help me with the below.

rahul@RGDesk:~/Downloads/rtl8192eu-linux-master$ sudo dkms install -m rtl8192eu  -v 1.0

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' all KVER=5.11.0-1015-raspi...(bad exit status: 2)
ERROR (dkms apport): binary package for rtl8192eu: 1.0 not found
Error! Bad return status for module build on kernel: 5.11.0-1015-raspi (aarch64)
Consult /var/lib/dkms/rtl8192eu/1.0/build/make.log for more information.

xvilen, 2021-08-18:
even if you somehow install its drivers it doesn't support monitor mode and packets injection

clnhub, 2021-08-19:
@Rahulgutal Could you try the clnhub_test5 branch for improved (32-bit ARM) support?

@xvilen clnhub_test4 has full monitor mode support, but is a bit behind.

Rahulgutal, 2021-08-28:
> @Rahulgutal Could you try the clnhub_test5 branch for improved (32-bit ARM) support?
> 
> @xvilen clnhub_test4 has full monitor mode support, but is a bit behind.

@clnhub i checked out the branch "clnhub_test5" for my DLINK usb adaptor on Raspberry pi 4 along with Ubuntu for Raspberry pi and i get the same error. 

Requesting help to proceed further logs are below:

root@RGDesk:/home/rahul/dlink_drivers/rtl8192eu-linux# rmmod 8192eu
rmmod: ERROR: Module 8192eu is not currently loaded
root@RGDesk:/home/rahul/dlink_drivers/rtl8192eu-linux# rmmod rtl8xxxu
rmmod: ERROR: Module rtl8xxxu is not currently loaded
root@RGDesk:/home/rahul/dlink_drivers/rtl8192eu-linux# dkms remove -m rtl8192eu -v 1.0
Error! The module/version combo: rtl8192eu-1.0
is not located in the DKMS tree.
root@RGDesk:/home/rahul/dlink_drivers/rtl8192eu-linux# echo "blacklist rtl8xxxu" >> ./blacklist-rtl8xxxu.conf
root@RGDesk:/home/rahul/dlink_drivers/rtl8192eu-linux# mv ./blacklist-rtl8xxxu.conf /etc/mod
modprobe.d/     modules         modules-load.d/
root@RGDesk:/home/rahul/dlink_drivers/rtl8192eu-linux# mv ./blacklist-rtl8xxxu.conf /etc/mod
modprobe.d/     modules         modules-load.d/
root@RGDesk:/home/rahul/dlink_drivers/rtl8192eu-linux# mv ./blacklist-rtl8xxxu.conf /etc/modprobe.d/
root@RGDesk:/home/rahul/dlink_drivers/rtl8192eu-linux# cp -ar . /usr/src/rtl8192eu-1.0
root@RGDesk:/home/rahul/dlink_drivers/rtl8192eu-linux# ^C
root@RGDesk:/home/rahul/dlink_drivers/rtl8192eu-linux# git branch
* clnhub_test5
  master
root@RGDesk:/home/rahul/dlink_drivers/rtl8192eu-linux# dkms add -m rtl8192eu -v 1.0

Creating symlink /var/lib/dkms/rtl8192eu/1.0/source ->
                 /usr/src/rtl8192eu-1.0

DKMS: add completed.
root@RGDesk:/home/rahul/dlink_drivers/rtl8192eu-linux# dkms install -m rtl8192eu -v 1.0

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' all KVER=5.11.0-1016-raspi...(bad exit status: 2)
ERROR (dkms apport): binary package for rtl8192eu: 1.0 not found
Error! Bad return status for module build on kernel: 5.11.0-1016-raspi (aarch64)
Consult /var/lib/dkms/rtl8192eu/1.0/build/make.log for more information.
root@RGDesk:/home/rahul/dlink_drivers/rtl8192eu-linux#

clnhub, 2021-08-28:
@Rahulgutal

Please try the master branch with the following changes (Makefile):

CONFIG_PLATFORM_I386_PC = n
CONFIG_PLATFORM_ARM_AARCH64 = y

hardikbvaghela, 2022-05-12:
i tried this in my ubuntu 18.4 but still its error....... what should i do?

~/rtl8192eu-linux$ sudo dkms install rtl8192eu/1.0

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' all KVER=4.9.253-tegra...(bad exit status: 2)
ERROR (dkms apport): binary package for rtl8192eu: 1.0 not found
Error! Bad return status for module build on kernel: 4.9.253-tegra (aarch64)
Consult /var/lib/dkms/rtl8192eu/1.0/build/make.log for more information.

dilekkrky, 2022-05-26:
Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' all KVER=4.9.253-tegra...(bad exit status: 2)
ERROR (dkms apport): binary package for rtl8192eu: 1.0 not found
Error! Bad return status for module build on kernel: 4.9.253-tegra (aarch64)
Consult /var/lib/dkms/rtl8192eu/1.0/build/make.log for more information.

Unfortunately i am getting same error.

closed 2025-01-02 by clnhub