clnhub/rtl8192eu-linux #15: AArch64 support

maxnatt, 2020-04-21:
Greetings, 

is there any way to compile this on AArch64 hardware? I have Raspberry Pi 4 with Fedora 32 (aarch64 build) and get following result when issuing `make`:

```sh
make ARCH=aarch64 CROSS_COMPILE= -C /lib/modules/5.6.2-301.fc32.aarch64/build M=/home/max/rtl8192eu-linux  modules
make[1]: Entering directory '/usr/src/kernels/5.6.2-301.fc32.aarch64'
Makefile:653: arch/aarch64/Makefile: No such file or directory
make[1]: *** No rule to make target 'arch/aarch64/Makefile'.  Stop.
make[1]: Leaving directory '/usr/src/kernels/5.6.2-301.fc32.aarch64'
make: *** [Makefile:1829: modules] Error 2
```

Is there any way to resolve the issue?

clnhub, 2020-04-22:
You can try adding 

` CONFIG_PLATFORM_ARM_RPI = y`

or

`CONFIG_PLATFORM_ARM_AARCH64=y`

to you Makefile and also set the default i386 platform to n.

You should probably also add the part below so the compiler can recognize your architecture.

```
 ifeq ($(CONFIG_PLATFORM_ARM_RPI), y)
 EXTRA_CFLAGS += -DCONFIG_LITTLE_ENDIAN
 EXTRA_CFLAGS += -DCONFIG_IOCTL_CFG80211 -DRTW_USE_CFG80211_STA_EVENT
 ARCH := arm
 CROSS_COMPILE :=
 KVER  := $(shell uname -r)
 KSRC ?= /lib/modules/$(KVER)/build
 MODDESTDIR := /lib/modules/$(KVER)/kernel/drivers/net/wireless/
 endif

 ifeq ($(CONFIG_PLATFORM_ARM_AARCH64), y)
 EXTRA_CFLAGS += -DCONFIG_LITTLE_ENDIAN
 ARCH := arm64
 CROSS_COMPILE :=
 KVER  := $(shell uname -r)
 KSRC ?= /lib/modules/$(KVER)/build
 MODDESTDIR := /lib/modules/$(KVER)/kernel/drivers/net/wireless/
 endif
```
 
Btw, I can't test if this works, because my primary environment is Ubuntu x64.

maxnatt, 2020-04-22:
Thanks for the suggestion, I'll try it in a few days and will write back.

maxnatt, 2020-05-01:
Sorry for the delay. I can confirm that providing `CONFIG_PLATFORM_ARM_AARCH64=y` and the snippet above works like a charm on Fedora 32 (aarch64) and RPi 4!

I haven't tested this driver in the long run yet, but at least I am able to connect to Wi-Fi, as stock driver provided in Fedora just gives timeouts when connecting.

closed 2020-05-01 by maxnatt