clnhub/rtl8192eu-linux #19: 'make' all KVER=5.6.14-olimex.....(bad exit status: 2)

gusantor, 2020-08-12:
Hello

I'm trying to make rtl8192eu driver

Now I'm using Olimex olinuxino A20 board with Debian buster image

#sudo uname -a
#Linux MyHost 5.6.14-olimex #164523 SMP Tue Jun 30 16:45:51 UTC 2020 armv7l GNU/Linux

After clone this repository and run

<code>
#sudo ./install_wifi.sh

rmmod: ERROR: Module 8192eu is not currently loaded
rmmod: ERROR: Module rtl8xxxu is not currently loaded

------------------------------
Deleting module version: 1.0
completely from the DKMS tree.
------------------------------
Done.

Creating symlink /var/lib/dkms/rtl8192eu/1.0/source ->
                 /usr/src/rtl8192eu-1.0

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' all KVER=5.6.14-olimex.....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.6.14-olimex (armv7l)
Consult /var/lib/dkms/rtl8192eu/1.0/build/make.log for more information.
modprobe: FATAL: Module 8192eu not found in directory /lib/modules/5.6.14-olimex
</code>

make log file says:

<code>
DKMS make.log for rtl8192eu-1.0 for kernel 5.6.14-olimex (armv7l)
mié ago 12 19:30:51 -04 2020
make ARCH=armv7l CROSS_COMPILE= -C /lib/modules/5.6.14-olimex/build M=/var/lib/dkms/rtl8192eu/1.0/build  modules
make[1]: Entering directory '/usr/src/linux-headers-5.6.14-olimex'
Makefile:653: arch/armv7l/Makefile: No such file or directory
make[1]: *** No rule to make target 'arch/armv7l/Makefile'.  Stop.
make[1]: Leaving directory '/usr/src/linux-headers-5.6.14-olimex'
make: *** [Makefile:1839: modules] Error 2
</code>

Please any advice to get this to run ?

Thanks

tootbrute, 2020-08-14:
I have the same issue. Looks like this is not made for anything other than i386 :(

```
uname -a
Linux nanopi-r2s 5.4.49-rockchip64 #20.05.7 SMP PREEMPT Sun Jun 28 18:17:52 UTC 2020 aarch64 GNU/Linux
```

```
/rtl8192eu-linux$ sudo dkms install rtl8192eu/1.0
Kernel preparation unnecessary for this kernel.  Skipping...
Building module:
cleaning build area...
'make' all KVER=5.4.49-rockchip64...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.4.49-rockchip64 (aarch64)
Consult /var/lib/dkms/rtl8192eu/1.0/build/make.log for more information.
```

```
/rtl8192eu-linux$ cat /var/lib/dkms/rtl8192eu/1.0/build/make.log
DKMS make.log for rtl8192eu-1.0 for kernel 5.4.49-rockchip64 (aarch64)
Fri 14 Aug 2020 01:33:53 AM UTC
make ARCH=aarch64 CROSS_COMPILE= -C /lib/modules/5.4.49-rockchip64/build M=/var/lib/dkms/rtl8192eu/1.0/build  modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.49-rockchip64'
Makefile:648: arch/aarch64/Makefile: No such file or directory
make[1]: *** No rule to make target 'arch/aarch64/Makefile'.  Stop.
make[1]: Leaving directory '/usr/src/linux-headers-5.4.49-rockchip64'
make: *** [Makefile:1839: modules] Error 2
```

tootbrute, 2020-08-14:
> Hello
> 
> I'm trying to make rtl8192eu driver
> 
> Now I'm using Olimex olinuxino A20 board with Debian buster image
> 
> #sudo uname -a
> #Linux MyHost 5.6.14-olimex #164523 SMP Tue Jun 30 16:45:51 UTC 2020 armv7l GNU/Linux
> 
> After clone this repository and run
> ` #sudo ./install_wifi.sh rmmod: ERROR: Module 8192eu is not currently loaded rmmod: ERROR: Module rtl8xxxu is not currently loaded ## Deleting module version: 1.0 completely from the DKMS tree. Done. Creating symlink /var/lib/dkms/rtl8192eu/1.0/source -> /usr/src/rtl8192eu-1.0 DKMS: add completed. Kernel preparation unnecessary for this kernel. Skipping... `
> 
> `Building module: cleaning build area... 'make' all KVER=5.6.14-olimex.....(bad exit status: 2) Error! Bad return status for module build on kernel: 5.6.14-olimex (armv7l) Consult /var/lib/dkms/rtl8192eu/1.0/build/make.log for more information. modprobe: FATAL: Module 8192eu not found in directory /lib/modules/5.6.14-olimex `
> 
> make log file says:
> ` DKMS make.log for rtl8192eu-1.0 for kernel 5.6.14-olimex (armv7l) mié ago 12 19:30:51 -04 2020 make ARCH=armv7l CROSS_COMPILE= -C /lib/modules/5.6.14-olimex/build M=/var/lib/dkms/rtl8192eu/1.0/build modules make[1]: Entering directory '/usr/src/linux-headers-5.6.14-olimex' Makefile:653: arch/armv7l/Makefile: No such file or directory make[1]: *** No rule to make target 'arch/armv7l/Makefile'. Stop. make[1]: Leaving directory '/usr/src/linux-headers-5.6.14-olimex' make: *** [Makefile:1839: modules] Error 2 `
> 
> Please any advice to get this to run ?
> 
> Thanks

try this one: https://github.com/Mange/rtl8192eu-linux-driver

i just did it right now and it worked.

clnhub, 2020-09-06:
Please try setting (Makefile)
```
CONFIG_PLATFORM_I386_PC = n
CONFIG_PLATFORM_ARM_AARCH64 = y
```
before installing.

gusantor, 2020-09-08:
Thanks for you help, sadly it doesn't work for me

I get response

'''
~/rtl8192eu-linux$ sudo ./install_wifi.sh 
rmmod: ERROR: Module 8192eu is not currently loaded
rmmod: ERROR: Module rtl8xxxu is not currently loaded

------------------------------
Deleting module version: 1.0
completely from the DKMS tree.
------------------------------
Done.

Creating symlink /var/lib/dkms/rtl8192eu/1.0/source ->
                 /usr/src/rtl8192eu-1.0

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' all KVER=5.6.14-olimex.....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.6.14-olimex (armv7l)
Consult /var/lib/dkms/rtl8192eu/1.0/build/make.log for more information.
modprobe: FATAL: Module 8192eu not found in directory /lib/modules/5.6.14-olimex
'''

log file says

'''
DKMS make.log for rtl8192eu-1.0 for kernel 5.6.14-olimex (armv7l)
mar sep  8 19:05:21 -03 2020
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/5.6.14-olimex/build M=/var/lib/dkms/rtl8192eu/1.0/build  modules
make[1]: Entering directory '/usr/src/linux-headers-5.6.14-olimex'
Makefile:653: arch/arm64/Makefile: No such file or directory
make[1]: *** No rule to make target 'arch/arm64/Makefile'.  Stop.
make[1]: Leaving directory '/usr/src/linux-headers-5.6.14-olimex'
make: *** [Makefile:1839: modules] Error 2
'''

Reading this, it appears there's a missing arch/arm64/Makefile file, bur really I don't know or understand how to solve

Any thoughts ? thanks

clnhub, 2020-09-10:
There seem to be multiple architectures involved here. The aarch64 GNU/Linux architecture should work with my Makefile comment.

The armv7l architecture is a different problem though, this is a 32-bit architecture. Could you try adding (below ###################### Platform Related #######################) :

`CONFIG_PLATFORM_ARM_RPI = y`
(and set the other architecture(s) to n)

and (below the `ifeq ($(CONFIG_PLATFORM_I386_PC), y)...` part:

```
ifeq ($(CONFIG_PLATFORM_ARM_RPI), y)
EXTRA_CFLAGS += -DCONFIG_LITTLE_ENDIAN
EXTRA_CFLAGS += -DCONFIG_IOCTL_CFG80211 -DRTW_USE_CFG80211_STA_EVENT
ARCH := arm
CROSS_COMPILE :=
KVER  := $(shell uname -r)
KSRC ?= /lib/modules/$(KVER)/build
MODDESTDIR := /lib/modules/$(KVER)/kernel/drivers/net/wireless/
endif
```

gusantor, 2020-09-10:
Thanks, fail again

I did changes you sugested

Received

```
~/rtl8192eu-linux$ sudo ./install_wifi.sh 
rmmod: ERROR: Module 8192eu is not currently loaded
rmmod: ERROR: Module rtl8xxxu is not currently loaded

------------------------------
Deleting module version: 1.0
completely from the DKMS tree.
------------------------------
Done.

Creating symlink /var/lib/dkms/rtl8192eu/1.0/source ->
                 /usr/src/rtl8192eu-1.0

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' all KVER=5.6.14-olimex..........(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.6.14-olimex (armv7l)
Consult /var/lib/dkms/rtl8192eu/1.0/build/make.log for more information.
modprobe: FATAL: Module 8192eu not found in directory /lib/modules/5.6.14-olimex
```

and log file

```
DKMS make.log for rtl8192eu-1.0 for kernel 5.6.14-olimex (armv7l)
jue sep 10 13:56:04 -03 2020
make ARCH=arm CROSS_COMPILE= -C /lib/modules/5.6.14-olimex/build M=/var/lib/dkms/rtl8192eu/1.0/build  modules
make[1]: Entering directory '/usr/src/linux-headers-5.6.14-olimex'
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_cmd.o
/bin/sh: 1: scripts/basic/fixdep: Exec format error
make[2]: *** [scripts/Makefile.build:268: /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_cmd.o] Error 2
make[2]: *** Deleting file '/var/lib/dkms/rtl8192eu/1.0/build/core/rtw_cmd.o'
make[1]: *** [Makefile:1691: /var/lib/dkms/rtl8192eu/1.0/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.6.14-olimex'
make: *** [Makefile:1854: modules] Error 2
```

Thank you

clnhub, 2020-09-10:
This looks like a kernel header issue caused by cross compiling. Did you try other drivers?

closed 2025-01-02 by clnhub