clnhub/rtl8192eu-linux #23: LED does not work

andreas-R17, 2020-09-30:
LED does not work on adapter MERCUSYS MW300UM.

`# lsusb`
`Bus 005 Device 002: ID 2c4e:0100`

How to make it work?

clnhub, 2020-10-07:
Try (MakeFile):
`EXTRA_CFLAGS += -DCONFIG_RTW_LED -DCONFIG_RTW_SW_LED -DCONFIG_LED_ENABLE`

andreas-R17, 2020-10-07:
Does not work((.

andreas-R17, 2020-10-07:
Thank you very much.

closed 2020-10-07 by clnhub

griloHBG, 2021-03-10:
This worked for TL-WN823N!

Thanks!

smdjll, 2021-11-12:
@griloHBG Sorry to bother you; I also have a TL-WN823N. Do you remember where exactly on the Makefile did you put the CFLAG? and which one worked out for you

clnhub, 2021-11-12:
Just add the whole line.

mraandtux, 2024-05-19:
>``` EXTRA_CFLAGS += -DCONFIG_RTW_LED -DCONFIG_RTW_SW_LED -DCONFIG_LED_ENABLE```

That line doesn't work on Ubuntu 24.04 with 6.8 kernel. From make.log:
```/var/lib/dkms/rtl8192eu/1.0/build/os_dep/linux/rtw_proc.c:640:5: error: no previous prototype for ‘proc_get_led_config’ [-Werror=missing-prototypes]```