clnhub/rtl8192eu-linux #26: Changing the network does not work,

TPLinux, 2020-12-08:


clnhub, 2020-12-08:
Hello,

Yeah that's a long standing issue that may need some further investigation. However, a quick fix is to reload the driver with the first and last line of the (install) script.

TPLinux, 2020-12-12:
> Hello,
> 
> Yeah that's a long standing issue that may need some further investigation. However, a quick fix is to reload the driver with the first and last line of the (install) script.

I will try it and response , ty

TPLinux, 2020-12-12:
> Hello,
> 
> Yeah that's a long standing issue that may need some further investigation. However, a quick fix is to reload the driver with the first and last line of the (install) script.

This workaround worked, thanks

closed 2020-12-12 by TPLinux