clnhub/rtl8192eu-linux #29: Doesn't compile on Ubuntu 20.04 with kernel 5.10.1

cyberdroid, 2020-12-19:
'make' all KVER=5.10.1-051001-generic...................................(bad exit status: 2)
ERROR (dkms apport): binary package for rtl8192eu: 1.0 not found
Error! Bad return status for module build on kernel: 5.10.1-051001-generic (x86_64)
Consult /var/lib/dkms/rtl8192eu/1.0/build/make.log for more information.
[make.log](https://github.com/clnhub/rtl8192eu-linux/files/5719693/make.log)

clnhub, 2020-12-19:
Thanks for the log file.

There is a bug in recent kernels (and Ubuntu) where module.lds is not added properly to the generic kernel directory. 

To (temporarily) fix this, please copy module.lds.S from /usr/src/linux-headers-5.10.1-051001/scripts to /usr/src/linux-headers-5.10.1-051001-generic/scripts, comment out the include part and rename it to module.lds.

It may be necessary to repeat the above part for every new 5.10(+) kernel until this is fixed by the kernel (compilation) team, because I do not want to add custom files to the kernel directory from the installer.

cyberdroid, 2020-12-19:
You wrote "comment out the include part and rename it to module.lds"
What include part?

clnhub, 2020-12-19:
#include <asm/module.lds.h>

clnhub, 2020-12-20:
[module_lds.sh](https://github.com/clnhub/rtl8192eu-linux/raw/clnhub_test3/module_lds.sh) is an automated version of the steps above. 

This should work if you run this on the 5.10.* kernel (from any directory) before (re)installing the driver.

clnhub, 2021-01-25:
(lds) Issues have been fixed by kernel/distribution maintainers.

closed 2021-01-25 by clnhub