clnhub/rtl8192eu-linux #31: Compiles, module loads but driver not working on Opensuse Tumbleweed Kernel 5.10.xx

srihari83, 2021-01-19:
Hello,
  The master branch compiles on Opensuse Tumbleweed using the installation script without any error/warning messages to the console. Module is created & also loads but the driver has not been working since kernel upgrades starting from 5.10.1. Present kernel version at 5.10.7 & still fails.
But make log shows many warnings.. these may be the cause. Have attached the make.log.. 

On the contrary, the old version of the driver 4.4.x on [this link](https://github.com/Mange/rtl8192eu-linux-driver) compiles, module loads & the driver works seamlessly. 
No particular error related to *.lds until now. 
Hope it helps.. Thanks a lot for the maintaining the repo until now!!
[clnhub_rtl8192eu_make.log](https://github.com/clnhub/rtl8192eu-linux/files/5838035/clnhub_rtl8192eu_make.log)

clnhub, 2021-01-23:
Hi,

There are no real issues in the (make) log file, just some smaller programming (style) warnings.

The default rtl8xxxu driver also doesn't work on openSUSE, while on Ubuntu it works @ about 1/10 of the normal speed.

Does it connect? Could you try debug mode and show the messages?

srihari83, 2021-01-23:
Hi,
 No, it doesn't connect. lsmod shows the module is loaded. Reloading after a depmod doesn't help.
 
I'm new to kernel/module debugging.. I tried the following:
        `cd /sys/module/8192eu/sections`
        `cat .text .data .bss`
  Output:
    `0xffffffffc137c000`
    `0xffffffffc1472000`
    `0xffffffffc149d380`
  
Then I unzipped the module file 8192eu.ko.xz & started gdb
  `gdb vmlinux /proc/kcore`
  `(gdb) add-symbol-file /lib/modules/5.10.7-1-default/kernel/drivers/net/wireless/8192eu.ko 0xffffffffc137c000 -s .data 0xffffffffc1472000 -s .bss 0xffffffffc149d380`
`add symbol table from file "/lib/modules/5.10.7-1-default/kernel/drivers/net/wireless/8192eu.ko" at`
	`.text_addr = 0xffffffffc137c000`
	`.data_addr = 0xffffffffc1472000`
	`.bss_addr = 0xffffffffc149d380`
`(y or n) y`
`Reading symbols from /lib/modules/5.10.7-1-default/kernel/drivers/net/wireless/8192eu.ko...`
`(No debugging symbols found in /lib/modules/5.10.7-1-default/kernel/drivers/net/wireless/8192eu.ko)`

I don't know if what tried here is useless or wrong.. Please let me know!
 
 
 In anycase, the output of `hwinfo --usb` shows:
 `07: USB 00.0: 0000 Unclassified device`
  `[Created at usb.122]`
  `.`
  `.`
  `Model: "D-Link Wireless N Nano USB Adapter"`
  `Hotplug: USB`
  `Vendor: usb 0x2001 "D-Link"`
  `Device: usb 0x3319 "Wireless N Nano USB Adapter"`
  `.`
  `.`
  `Driver Info #0:`
    `Driver Status: rtl8xxxu is not active`
    `Driver Activation Cmd: "modprobe rtl8xxxu"`
  `.`
  `.`
  
I unloaded & reloaded the module in verbose. Output of `modprobe -v 8192eu` shows:
`insmod /lib/modules/5.10.7-1-default/kernel/drivers/net/wireless/8192eu.ko.xz `

When I compile the older version 4.4.x from the Mange repo on this [link](https://github.com/Mange/rtl8192eu-linux-driver), the make.log has no warnings. The module loads & the wifi works. Though the connect is a little slower. 
The output of `hwinfo --usb` with driver version 4.4.x shows:

 `07: USB 00.0: 0200 Ethernet controller`
  `[Created at usb.122]`
  `.`
  `.`
  `Model: "D-Link Wireless N Nano USB Adapter"`
  `Hotplug: USB`
  `Vendor: usb 0x2001 "D-Link"`
  `Device: usb 0x3319 "Wireless N Nano USB Adapter"`
  `.`
  `.`
  `Driver Info #0:`
    `Driver Status: 8192eu is active`
    `Driver Activation Cmd: "modprobe 8192eu"`
  `.`
  `.`

srihari83, 2021-01-25:
Hi! 
Latest push compatible with kernel 5.11 works perfectly! Thanks a lot for the update! 
Please close this issue..

clnhub, 2021-01-25:
Ok, thanks for the detailed report!

closed 2021-01-25 by clnhub