clnhub/rtl8192eu-linux #32: No packet insertion

harlequin2, 2021-01-21:
Just installed in Kali 5.9 and it starts and stops monitor mode OK but no insertion. airmon-ng shows Chipset TP-Link TL-WN822N Version 4 RTL8192EU but Model on the unit's sticker shows WN822N(US) Ver 5.20.
aireplay-ng -9 wlan0 gives:
2:38:55  Trying broadcast probe requests...
12:38:57  No Answer...
12:38:57  Found 2 APs

12:38:57  Trying directed probe requests...
12:38:57  88:D7:F6:AA:A6:08 - channel: 6 - 'Nevilles'
12:39:03   0/30:   0%
"Nevilles" is my home network.
How to get insertion working?

clnhub, 2021-01-22:
Did you enable monitor mode in the make script?

harlequin2, 2021-01-22:
> 
> 
> Did you enable monitor mode in the make script?

No, I just ran it as-is. But the interface goes into monitor mode OK ie airmon-ng start and stop work fine and airodump-ng shows packet activity.

clnhub, 2021-01-22:
Please try (Makefile)
`CONFIG_WIFI_MONITOR = y`
and re-install

harlequin2, 2021-01-23:
> 
> 
> Please try (Makefile)
> `CONFIG_WIFI_MONITOR = y`
> and re-install

I did that and found no difference. After looking through the Makefile, I found it already had CONFIG_WIFI_MONITOR = y so that's why. Any other suggestions?

clnhub, 2021-01-28:
You can try the clnhub_test4 branch.

albaluz35, 2024-05-16:
In my case enabling monitor mode and doing packet injection WORKS! as follows:

sudo apt update
sudo apt upgrade
sudo apt-get install linux-headers-`uname -r` build-essential dkms
git clone https://github.com/clnhub/rtl8192eu-linux.git
cd rtl8192eu-linux
sed -i -E 's/CONFIG_WIFI_MONITOR\s*=\s*n/CONFIG_WIFI_MONITOR = y/' Makefile

sudo rmmod 8192eu rtl8xxxu r8188eu
echo "blacklist rtl8xxxu" >> ./blacklist-rtl8xxxu.conf
sudo mv ./blacklist-rtl8xxxu.conf /etc/modprobe.d/
./install_wifi.sh

-----------------------

**IF THERE ARE PROBLEMS WITH ./install_wifi.sh, RUN ./uninstall_wifi.sh then execute make commands manually:**
\# sudo make
\# sudo make install

-----------------

sudo modprobe 8192eu
sudo lsmod |grep 8192eu                                                     

!To enable Monitor mode and test packet injection:!
=================================================

WL=wlan0
sudo iwconfig
sudo ifconfig $WL down
sudo airmon-ng check kill
sudo iwconfig $WL mode monitor
sudo ifconfig $WL up
sudo iwconfig

sudo aireplay-ng --test $WL

closed 2025-01-02 by clnhub