clnhub/rtl8192eu-linux #35: My system crashes 5.11.2-arch1-1/TL-WN822N

jaderebrasil, 2021-03-04:
I tried hard but I have no idea what to do.

I have TL-WN822N and I was using this driver for quite a while. But recently, since I reinstalled my Arch Linux,
now with 5.11.2-arch1-1 kernel, when I use load 8192eu module, everything freezes.

ip a, dkms remove, rmmod, nothing related to the network works. 

When I try to run something, everything is stopped and even waiting a lot, nothing happens.

I don't know much about it, but I think it may be something with the kernel that is incompatible. I don't know if anyone has tried to run this driver on 5.11.

eugeneBecker, 2021-03-05:
Yeah. Driver stop working in mainstream kernel, but it still working on LTS kernel.

eugeneBecker, 2021-03-05:
> I tried hard but I have no idea what to do.
>
> I have TL-WN822N and I was using this driver for quite a while. But recently, since I reinstalled my Arch Linux,
> now with 5.11.2-arch1-1 kernel, when I [...]

eugeneBecker, 2021-03-05:
> I tried hard but I have no idea what to do.
>
> I have TL-WN822N and I was using this driver for quite a while. But recently, since I reinstalled my Arch Linux,
> now with 5.11.2-arch1-1 kernel, when I [...]

Just install LTS kernel.
sudo pacman -S linux-lts linux-lts-headers

jaderebrasil, 2021-03-05:
> Just install LTS kernel.
> sudo pacman -S linux-lts linux-lts-headers

Oh, I didn't even think about it. Thanks! This worked perfectly.

closed 2021-03-05 by jaderebrasil