clnhub/rtl8192eu-linux #37: Installation error on Kali 5.10

danieljs777, 2021-03-23:
./install_wifi.sh 
rmmod: ERROR: Module 8192eu is not currently loaded
rmmod: ERROR: Module rtl8xxxu is not currently loaded
Error! The module/version combo: rtl8192eu-1.0
is not located in the DKMS tree.

Creating symlink /var/lib/dkms/rtl8192eu/1.0/source ->
                 /usr/src/rtl8192eu-1.0

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' all KVER=5.10.0-kali4-amd64........(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.10.0-kali4-amd64 (x86_64)

in1rud3r, 2021-03-23:
First make sure to update Kali to the latest version : 
`sudo apt update`
`sudo apt full-upgrade -y`
Make sure you have the prerequisites installed : 
`sudo apt -y install linux-headers-generic build-essential dkms git`
I had some issues with the script too, try doing it without the script (manually), fixed it for me :
`sudo dkms add .
sudo dkms install rtl8192eu/1.0
sudo modprobe 8192eu`

starkchaudhary, 2021-03-30:
> ./install_wifi.sh
> rmmod: ERROR: Module 8192eu is not currently loaded
> rmmod: ERROR: Module rtl8xxxu is not currently loaded
> Error! The module/version combo: rtl8192eu-1.0
> is not located in the DKMS [...]

I Solved it same problem in mine
kali 5.10.0

just type there 
make clean
make 
make install and by typing these commands simply it will install 
then type
 sudo modprobe 8192eu

and your adapter is good to go

danieljs777, 2021-04-06:
Hi guys! I already tried that with other drivers, but reviewing those steps on this release has come to work. Thanks!!

closed 2021-04-06 by danieljs777