clnhub/rtl8192eu-linux #38: Makes the system unusable in 5.11.7-200.fc33.x86_64

ntrrgc, 2021-03-23:
This issue persists even after updating to the latest commit in the repo and reinstalling.

When running in kernel 5.11.7-200.fc33.x86_64 in Fedora the system becomes unusable the moment the USB Wi-Fi card is plugged. Basic commands such as `sudo`, `ip` or `iwconfig` stall forever with no output. So do all processes that do basic operations with the network, becoming unresponsive even to SIGKILL.

clnhub, 2021-03-23:
The latest (master) commit did solve the issue in Ubuntu. Can you try the 5.6.3.1 branch?

ntrrgc, 2021-03-24:
I just tried 5.6.3.1, same issue.

vstyler96, 2023-01-12:
This should have more attention, this same thing happens to me, on Arch Linux

clnhub, 2023-01-12:
Is this with 5.11.2.1? Could you try a more recent kernel?

vstyler96, 2023-01-12:
> Is this with 5.11.2.1? Could you try a more recent kernel?

The only version working with the Linux kernel 6.1.4 is master version (v5.6.3.1). But the the speed of wifi is excessively less than in windows.

OS: Arch Linux
Driver version: master (5.6.3.1)
Kernel Version: 6.1.4
Device: TP-LINK TL-WN8200ND(UN) v3.0

clnhub, 2023-01-12:
It is hard to debug this way. I have had no issues with the current branch 5.11.2.1 with all kernels starting with 5.8 (up to 6.2) on multiple distro's.

Could you try a different config like Kali (live)? It comes with development packages so you can run the driver installer (ram drive) directly after boot. Also, could you try kernel 5.10 on Arch to be sure?

closed 2025-01-02 by clnhub