clnhub/rtl8192eu-linux #40: Can't compile module rtl8192eu because `GLIBC_2.33' not found

cyberdroid, 2021-07-16:
[make.log](https://github.com/clnhub/rtl8192eu-linux/files/6827940/make.log)

cyberdroid, 2021-07-16:
Ubuntu 20.04

clnhub, 2021-07-16:
I assume you are using a mainline kernel?

Mainline kernels are now built on the newest Ubuntu version with glibc v2.33 and thus kernel modules require the same version. From what I know 'glibc' is part of the distribution and is hard to update.

This is the bug report about this.

[https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1926938](https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1926938)

You can wait for an official fix, or:

1. Try to upgrade glibc to v2.33
2. Upgrade to v21.04
3. Use the official Ubuntu kernel

CGarces, 2021-07-16:
This should work to compile the kernel but maybe broke something, use at your own risk.

```bash
wget -nv http://mirrors.kernel.org/ubuntu/pool/main/g/glibc/libc6_2.33-0ubuntu5_amd64.deb
sudo dpkg --force-all -i libc6_2.33-0ubuntu5_amd64.deb
```

Works fine on my CI builds
https://github.com/Mange/rtl8192eu-linux-driver/commit/4013a01fe6e0a6e34853f9d2885d3cb02f5f230e

cyberdroid, 2021-07-16:
Thanks for the responses.
Ok, now the problem is clear.
I think the best solution will be installing ubuntu 21.04 in a VM and compile rtl8192eu  module there then copy that module to 20.04 kernel tree.

jameshilliard, 2022-10-14:
> scripts/basic/fixdep: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by scripts/basic/fixdep)

This indicates the userspace utility [`scripts/basic/fixdep`](https://github.com/torvalds/linux/blob/v5.12/tools/build/fixdep.c) was linked against a newer glibc, something probably messed up in your build env.

> Mainline kernels are now built on the newest Ubuntu version with glibc v2.33 and thus kernel modules require the same version. From what I know 'glibc' is part of the distribution and is hard to update.

This doesn't make sense, libc(ie glibc) is a usespace thing, kernel modules don't use a libc at all.

clnhub, 2022-10-14:
Yep, modpost and fixdep. This is only valid for Ubuntu's pre-compiled kernels. Compiling from source always works.