clnhub/rtl8192eu-linux #43: Not working for ubuntu 20.04 kernel  version 5.11

Leonardo-LX, 2021-11-17:
I update my system from ubuntu16.04 to ubuntu 20.04 
kernel version is 5.11.
I installed the rtl8192eu driver, but network-manager always show 'not adapter', is rtl8192eu driver not  suitable for ubuntu20.04?or else?

clnhub, 2021-11-17:
Which version (branch) did you try? (Master/5.6.3.1/5.6.4)?

Did it compile without issues? If it didn't, please post (make) log, if it did, please post `lsusb` output.

Vanyoo, 2021-11-22:
Same system, using `./install_wifi.sh` will get the last print:
```
DKMS: install completed.
modprobe: ERROR: could not insert '8192eu': Exec format error
```
And `dmesg` will shower: 
```
[ 1377.894086] usbcore: registered new interface driver rtl8xxxu
[ 1921.189009] usbcore: deregistering interface driver rtl8xxxu
[ 2026.874541] Module has invalid ELF structures
```

`lsusb` shows:
```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 1c4f:0034 SiGma Micro Usb Mouse
Bus 001 Device 003: ID 2357:0135 TP-Link 802.11n  WLAN Adapter
Bus 001 Device 002: ID 2a7a:8a47  CASUE USB KB
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Help. Dispute me several days.

clnhub, 2021-11-22:
Could you post kernel and module c++ version?

`cat /proc/version`
`gcc --version`

Vanyoo, 2021-11-22:
`cat /proc/version`:
```
Linux version 5.11.0-40-generic (buildd@lgw01-amd64-010) (gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #44~20.04.2-Ubuntu SMP Tue Oct 26 18:07:44 UTC 2021
```

`gcc --version`:
```
gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Thanks for ur reply.

clnhub, 2021-11-22:
Can you run some tests on the generate module (8192eu.ko).

`/var/lib/dkms/rtl8192eu/1.0/<kernel version>/x86_64/module/8192eu.ko`

Try:

`readelf -p .comment 8192eu.ko`
`readelf -h 8192eu.ko`

Also, please compile (without installing, don't mind the compiler warnings!) from within the driver directory with:

`make clean`
`make`

And also run the readelf tests on the generated 8192eu.ko file.

Does it work with other kernels?

Vanyoo, 2021-11-23:
`readelf -p .comment 8192eu.ko` prints:
```

String dump of section '.comment':
  [     1]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [    2c]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [    57]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [    82]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [    ad]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [    d8]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   103]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   12e]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   159]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   184]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   1af]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   1da]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   205]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   230]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   25b]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   286]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   2b1]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   2dc]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   307]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   332]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   35d]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   388]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   3b3]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   3de]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   409]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   434]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   45f]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   48a]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   4b5]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   4e0]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   50b]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   536]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   561]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   58c]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   5b7]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   5e2]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   60d]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   638]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   663]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   68e]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   6b9]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   6e4]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   70f]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   73a]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   765]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   790]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   7bb]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   7e6]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   811]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   83c]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   867]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   892]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   8bd]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   8e8]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   913]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   93e]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   969]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   994]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   9bf]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   9ea]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   a15]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   a40]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   a6b]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   a96]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   ac1]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
  [   aec]  GCC: (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
[... 65 lines cut ...]
```

`readelf -h 8192eu.ko` prints:
```
ELF Header:
  Magic:   7f 45 4c 46 02 01 01 00 00 00 00 00 00 00 00 00
  Class:                             ELF64
  Data:                              2's complement, little endian
  Version:                           1 (current)
  OS/ABI:                            UNIX - System V
  ABI Version:                       0
  Type:                              REL (Relocatable file)
  Machine:                           Advanced Micro Devices X86-64
  Version:                           0x1
  Entry point address:               0x0
  Start of program headers:          0 (bytes into file)
  Start of section headers:          2262616 (bytes into file)
  Flags:                             0x0
  Size of this header:               64 (bytes)
  Size of program headers:           0 (bytes)
  Number of program headers:         0
  Size of section headers:           64 (bytes)
  Number of section headers:         41
  Section header string table index: 40
```

clnhub, 2021-11-23:
I have successfully tried the driver on 20.04 with kernel 5.11, so this seems to be caused by upgrading?

Can you re-install the driver, but replace the dkms uninstall/remove lines with (instead of -m -v)

```
sudo dkms remove rtl8192eu/1.0
sudo dkms uninstall rtl8192eu/1.0
```

Also If that doesn't work, can you try `modprobe --force 8192eu` and post `dmesg | tail`. There must be a more detailed error message about which (old) library, or header (link?) is causing this, because there does not seem to be anything wrong from the driver side.

Vanyoo, 2021-11-24:
```
[    0.000000] microcode: microcode updated early to revision 0xea, date = 2021-01-05
[    0.000000] Linux version 5.11.0-40-generic (buildd@lgw01-amd64-010) (gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #44~20.04.2-Ubuntu SMP Tue Oct 26 18:07:44 UTC 2021 (Ubuntu 5.11.0-40.44~20.04.2-generic 5.11.22)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-5.11.0-40-generic root=UUID=9c5d11b8-bb68-47be-b8af-eb27a382a99f ro quiet splash vt.handoff=7
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Hygon HygonGenuine
[    0.000000]   Centaur CentaurHauls
[    0.000000]   zhaoxin   Shanghai
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: xstate_offset[3]:  832, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]:  896, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Enabled xstate features 0x1f, context size is 960 bytes, using 'compacted' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000005efff] usable
[    0.000000] BIOS-e820: [mem 0x000000000005f000-0x000000000005ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000060000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000003fffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000040000000-0x00000000403fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000040400000-0x00000000824abfff] usable
[    0.000000] BIOS-e820: [mem 0x00000000824ac000-0x00000000824acfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000824ad000-0x00000000824adfff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000824ae000-0x000000008990efff] usable
[    0.000000] BIOS-e820: [mem 0x000000008990f000-0x0000000089d8efff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000089d8f000-0x0000000089f6bfff] usable
[    0.000000] BIOS-e820: [mem 0x0000000089f6c000-0x000000008a31efff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000008a31f000-0x000000008aefefff] reserved
[    0.000000] BIOS-e820: [mem 0x000000008aeff000-0x000000008aefffff] usable
[    0.000000] BIOS-e820: [mem 0x000000008af00000-0x000000008fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe000000-0x00000000fe010fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000026dffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] e820: update [mem 0x81c73018-0x81c83057] usable ==> usable
[    0.000000] e820: update [mem 0x81c73018-0x81c83057] usable ==> usable
[    0.000000] e820: update [mem 0x81c62018-0x81c72e57] usable ==> usable
[    0.000000] e820: update [mem 0x81c62018-0x81c72e57] usable ==> usable
[    0.000000] extended physical RAM map:
[    0.000000] reserve setup_data: [mem 0x0000000000000000-0x000000000005efff] usable
[    0.000000] reserve setup_data: [mem 0x000000000005f000-0x000000000005ffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000060000-0x000000000009ffff] usable
[    0.000000] reserve setup_data: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000100000-0x000000003fffffff] usable
[    0.000000] reserve setup_data: [mem 0x0000000040000000-0x00000000403fffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000040400000-0x0000000081c62017] usable
[    0.000000] reserve setup_data: [mem 0x0000000081c62018-0x0000000081c72e57] usable
[    0.000000] reserve setup_data: [mem 0x0000000081c72e58-0x0000000081c73017] usable
[    0.000000] reserve setup_data: [mem 0x0000000081c73018-0x0000000081c83057] usable
[    0.000000] reserve setup_data: [mem 0x0000000081c83058-0x00000000824abfff] usable
[    0.000000] reserve setup_data: [mem 0x00000000824ac000-0x00000000824acfff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x00000000824ad000-0x00000000824adfff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000824ae000-0x000000008990efff] usable
[    0.000000] reserve setup_data: [mem 0x000000008990f000-0x0000000089d8efff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000089d8f000-0x0000000089f6bfff] usable
[    0.000000] reserve setup_data: [mem 0x0000000089f6c000-0x000000008a31efff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x000000008a31f000-0x000000008aefefff] reserved
[    0.000000] reserve setup_data: [mem 0x000000008aeff000-0x000000008aefffff] usable
[    0.000000] reserve setup_data: [mem 0x000000008af00000-0x000000008fffffff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fe000000-0x00000000fe010fff] reserved
[... cut ...] arser"
[129301.185517] audit: type=1400 audit(1637683241.763:71): apparmor="DENIED" operation="capable" profile="/usr/sbin/cups-browsed" pid=53010 comm="cups-browsed" capability=23  capname="sys_nice"
```

I used `sudo dkms remove rtl8192eu/1.0   sudo dkms uninstall rtl8192eu/1.0` in `install_wifi.sh`, add `--force` in the script too, add `dmesg | tail` at last. 
I am unfamiliar with the `dmesg`, cannot find the detail error message about which library or header causing this...So I post the all.

clnhub, 2021-11-24:
You can try `ldconfig -v` or `ldconfig -X -N -v` (normal user). This will check and/or rebuild library cache and links.

Vanyoo, 2021-11-24:
Terrible...I tried, and not have any error. Then I tried `modprobe --force 8192eu` and `install_wifi.sh`, but still has this error :  `modprobe: ERROR: could not insert '8192eu': Exec format error`

clnhub, 2021-11-24:
Yeah that's not good. Did ldconfig not show any errors at all? I have had no issues with the driver on 20.04, 20.10, 21.04 and 21.10 (current) with any 5.x kernel other than the (fixed) issues mentioned in the commits.

The code is fine, there is something not linking correctly on your system. This could be corruption, missing config files, links, or headers. It is hard to debug if nothing else fails.

Can you run (and post) `modinfo 8192eu.ko` ? And also run `modprobe -v 8192eu.ko` ? There must be more information available.

Vanyoo, 2021-11-25:
` modinfo 8192eu.ko`:
```
filename:       /home/chafou/Desktop/rtl8192eu-linux-master/8192eu.ko
version:        v5.2.19.1_25633.20171222_COEX20171113-0047
author:         Realtek Semiconductor Corp.
description:    Realtek Wireless Lan Driver
license:        GPL
srcversion:     E2FD491C8080E7FD6059913
alias:          usb:v2C4Ep0104d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2C4Ep0100d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p0126d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p0109d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p0107d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p0108d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p3319d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p3312d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp818Cd*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v0BDAp818Bd*dc*dsc*dp*icFFiscFFipFFin*
depends:        cfg80211
retpoline:      Y
name:           8192eu
vermagic:       5.11.0-40-generic SMP mod_unload modversions
parm:           rtw_ips_mode:The default IPS mode (int)
parm:           rtw_lps_level:The default LPS level (int)
parm:           rtw_usb_rxagg_mode:int
parm:           rtw_dynamic_agg_enable:int
parm:           rtw_drv_log_level:set log level when insert driver module, default log level is _DRV_INFO_ = 4 (uint)
parm:           rtw_tx_bw_mode:The max tx bw for 2.4G and 5G. format is the same as rtw_bw_mode (uint)
parm:           rtw_rx_ampdu_sz_limit_1ss:RX AMPDU size limit for 1SS link of each BW, 0xFF: no limitation (array of uint)
parm:           rtw_rx_ampdu_sz_limit_2ss:RX AMPDU size limit for 2SS link of each BW, 0xFF: no limitation (array of uint)
parm:           rtw_rx_ampdu_sz_limit_3ss:RX AMPDU size limit for 3SS link of each BW, 0xFF: no limitation (array of uint)
parm:           rtw_rx_ampdu_sz_limit_4ss:RX AMPDU size limit for 4SS link of each BW, 0xFF: no limitation (array of uint)
parm:           rtw_rf_config:int
parm:           rtw_country_code:The default country code (in alpha2) (charp)
parm:           rtw_channel_plan:The default chplan ID when rtw_alpha2 is not specified or valid (int)
parm:           rtw_excl_chs:exclusive channel array (array of uint)
parm:           rtw_btcoex_enable:BT co-existence on/off, 0:off, 1:on, 2:by efuse (int)
parm:           rtw_ant_num:Antenna number setting, 0:by efuse (int)
parm:           rtw_qos_opt_enable:int
parm:           ifname:The default name to allocate for first interface (charp)
parm:           if2name:The default name to allocate for second interface (charp)
parm:           rtw_pwrtrim_enable:int
parm:           rtw_initmac:charp
parm:           rtw_special_rf_path:int
parm:           rtw_chip_version:int
parm:           rtw_rfintfs:int
parm:           rtw_lbkmode:int
parm:           rtw_network_mode:int
parm:           rtw_channel:int
parm:           rtw_mp_mode:int
parm:           rtw_wmm_enable:int
parm:           rtw_vrtl_carrier_sense:int
parm:           rtw_vcs_type:int
parm:           rtw_busy_thresh:int
parm:           rtw_ht_enable:int
parm:           rtw_bw_mode:int
parm:           rtw_ampdu_enable:int
parm:           rtw_rx_stbc:int
parm:           rtw_rx_ampdu_amsdu:int
parm:           rtw_tx_ampdu_amsdu:int
parm:           rtw_lowrate_two_xmit:int
parm:           rtw_power_mgnt:int
parm:           rtw_smart_ps:int
parm:           rtw_low_power:int
parm:           rtw_wifi_spec:int
parm:           rtw_full_ch_in_p2p_handshake:int
parm:           rtw_antdiv_cfg:int
parm:           rtw_antdiv_type:int
parm:           rtw_drv_ant_band_switch:int
parm:           rtw_single_ant_path:int
[... 27 lines cut ...]
parm:           rtw_rxgain_offset_5gl:default RF Gain 5GL Offset value:0 (uint)
parm:           rtw_rxgain_offset_5gh:uint
parm:           rtw_rxgain_offset_5gm:default RF Gain 5GM Offset value:0 (uint)
parm:           rtw_pll_ref_clk_sel:force pll_ref_clk_sel, 0xF:use autoload value (uint)
parm:           rtw_tx_pwr_by_rate:0:Disable, 1:Enable, 2: Depend on efuse (int)
parm:           rtw_target_tx_pwr_2g_a:2.4G target tx power (unit:dBm) of RF path A for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_2g_b:2.4G target tx power (unit:dBm) of RF path B for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_2g_c:2.4G target tx power (unit:dBm) of RF path C for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_2g_d:2.4G target tx power (unit:dBm) of RF path D for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_phy_file_path:The path of phy parameter (charp)
parm:           rtw_load_phy_file:PHY File Bit Map (int)
parm:           rtw_decrypt_phy_file:Enable Decrypt PHY File (int)
parm:           rtw_en_napi:int
parm:           rtw_en_gro:int
parm:           rtw_iqk_fw_offload:int
```

`modprobe -v 8192eu.ko`:
```
modprobe: FATAL: Module 8192eu.ko not found in directory /lib/modules/5.11.0-40-generic
```

The second command, I also tried `modprobe -v ./8192eu.ko` in the build directory, but it still find in ` /lib/modules/5.11.0-40-generic`.

I am really appreciate your patient help. Maybe I should buy one latest adaptor to resolve this problem, LOL, really cost ur long time.

clnhub, 2021-11-25:
Fortunately, these are basic Linux developer troubleshooting steps, which do not take a lot of time. Linux seems to be even worse than Windows with keeping track of header and library changes between updates.

Another step is to remove and re-install the build software.

` sudo apt-get purge linux-headers-generic build-essential dkms git`

Remove the kernel header directories in /usr/src/ for you current kernel (5.11) only and also remove the rtl8192eu-1.0 directory.

`sudo rm -r .usr/src/linux-headers-5.11.0-40-generic`
`sudo rm -r .usr/src/linux-headers-5.11.0-40`
`sudo rm -r /usr/src/rtl8192eu-1.0`

Reboot and re-install the build software (this requires an active internet connection, so you may need to restore rtl8xxxu temporarily with running `sudo modprobe rtl8xxxu`)

` sudo apt-get install linux-headers-generic build-essential dkms git`

Run the installer again.

Hansander, 2022-01-12:
Cloning the repo and running ./install_wifi worked like a charm, thank you.  I am on Manjaro kernel 5.15.

closed 2025-01-02 by clnhub