clnhub/rtl8192eu-linux #45: Not Loading for Kernel 5.17 in Fedora Rawhide

onyeibo, 2022-02-03:
This driver works for kernel 5.16 (although with poor performance).  Kernel 5.17 hit rawhide repos and the driver compiled successfully.  However, the driver does not load.

`nmcli device wifi list`

The above lists not device in 5.17 but when I boot with 5.16 the device is present.

sunwire, 2022-02-03:
Did you try [this branch](https://github.com/clnhub/rtl8192eu-linux/tree/5.6.4)?

onyeibo, 2022-02-04:
> Did you try [this branch](https://github.com/clnhub/rtl8192eu-linux/tree/5.6.4)?

Yes I did.

Here is the terminal output:

```
rmmod: ERROR: Module 8192eu is not currently loaded
rmmod: ERROR: Module rtl8xxxu is not currently loaded
Module rtl8192eu 1.0 is not installed for kernel 5.17.0-0.rc2.83.fc36.x86_64 (x86_64). Skipping...
Module rtl8192eu 1.0 is not installed for kernel 5.17.0-0.rc2.83.fc36.x86_64 (x86_64). Skipping...
Module rtl8192eu 1.0 is not built for kernel 5.17.0-0.rc2.83.fc36.x86_64 (x86_64). Skipping...
Deprecated feature: REMAKE_INITRD
Creating symlink /var/lib/dkms/rtl8192eu/1.0/source -> /usr/src/rtl8192eu-1.0
Deprecated feature: REMAKE_INITRD

Building module:
cleaning build area...
'make' all KVER=5.17.0-0.rc2.83.fc36.x86_64...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.17.0-0.rc2.83.fc36.x86_64 (x86_64)
Consult /var/lib/dkms/rtl8192eu/1.0/build/make.log for more information.
modprobe: FATAL: Module 8192eu not found in directory /lib/modules/5.17.0-0.rc2.83.fc36.x86_64
```

The contents of "/var/lib/dkms/rtl8192eu/1.0/build/make.log" follows:

```
DKMS make.log for rtl8192eu-1.0 for kernel 5.17.0-0.rc2.83.fc36.x86_64 (x86_64)
Fri 04 Feb 2022 05:03:23 WAT
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.17.0-0.rc2.83.fc36.x86_64/build M=/var/lib/dkms/rtl8192eu/1.0/build  modules
make[1]: Entering directory '/usr/src/kernels/5.17.0-0.rc2.83.fc36.x86_64'
/bin/sh: -c: line 1: syntax error near unexpected token `('
/bin/sh: -c: line 1: `if [ "gcc (GCC) 12.0.1 20220125 (Red Hat 12.0.1-0)" != ""gcc (GCC) 12.0.1 20220129 (Red Hat 12.0.1-0)"" ]; then \'
make[1]: *** [Makefile:1727: prepare] Error 2
make[1]: Leaving directory '/usr/src/kernels/5.17.0-0.rc2.83.fc36.x86_64'
make: *** [Makefile:2257: modules] Error 2
```

sunwire, 2022-02-04:
Would you like to test [this package](https://download.copr.fedorainfracloud.org/results/sunwire/dkms-rtl8192eu/fedora-rawhide-x86_64/03294042-dkms-rtl8192eu/dkms-rtl8192eu-5.6.4-1.git20220202.fc36.src.rpm)?

onyeibo, 2022-02-04:
> Would you like to test [this package](https://download.copr.fedorainfracloud.org/results/sunwire/dkms-rtl8192eu/fedora-rawhide-x86_64/03294042-dkms-rtl8192eu/dkms-rtl8192eu-5.6.4-1.git20220202.fc36.src.rpm)?

Done!

The src package compiles and installs fine but the module still does not load.

`lsmod | grep 8192eu`

The above outputs nothing.

onyeibo, 2022-02-04:
> /bin/sh: -c: line 1: `if [ "gcc (GCC) 12.0.1 20220125 (Red Hat 12.0.1-0)" != ""gcc (GCC) 12.0.1 20220129 (Red Hat 12.0.1-0)"" ]; then \'

Which file has that line?  There is a double quote on the right side of the comparison.   Could that be the issue?

sunwire, 2022-02-04:
> Which file has that line? There is a double quote on the right side of the comparison. Could that be the issue?

It's /usr/src/kernels/5.17.0-0.rc2.83.fc36.x86_64/Makefile from kernel-devel
And I think [this patch](https://www.spinics.net/lists/kernel/msg4179089.html) is to blame for it 
```
diff --git a/Makefile b/Makefile
index 4e8ac0730f51..aa4e5dc12049 100644
--- a/Makefile
+++ b/Makefile
@@ -1729,9 +1729,9 @@ PHONY += prepare
 # now expand this into a simple variable to reduce the cost of shell evaluations
 prepare: CC_VERSION_TEXT := $(CC_VERSION_TEXT)
 prepare:
-	@if [ "$(CC_VERSION_TEXT)" != $(CONFIG_CC_VERSION_TEXT) ]; then \
+	@if [ "$(CC_VERSION_TEXT)" != "$(CONFIG_CC_VERSION_TEXT)" ]; then \
 		echo >&2 "warning: the compiler differs from the one used to build the kernel"; \
-		echo >&2 "  The kernel was built by: "$(CONFIG_CC_VERSION_TEXT); \
+		echo >&2 "  The kernel was built by: $(CONFIG_CC_VERSION_TEXT)"; \
 		echo >&2 "  You are using:           $(CC_VERSION_TEXT)"; \
 	fi
 ```
If you undo these changes it compiles without problem

onyeibo, 2022-02-04:
>If you undo these changes it compiles without problem

Then perhaps I should file a bug against kernel-devel?  Not sure the kernel developers will be keen on reverting their new quoting rules for one wifi driver.

sunwire, 2022-02-04:
> Then perhaps I should file a bug against kernel-devel? Not sure the kernel developers will be keen on reverting their new quoting rules for one wifi driver.

I think you should. First try it on https://bugzilla.redhat.com/ 
It's not one wifi driver every dkms package will fail.

onyeibo, 2022-02-05:
[Bug Report](https://bugzilla.redhat.com/show_bug.cgi?id=2050944) filed at Bugzilla

onyeibo, 2022-02-09:
Issue appear to be fixed in newer versions of kernel 5.17.   I'm glad its fixed.  It remains to get better performance from the driver/dongle but that's a different issue.   Closing this Issue now

closed 2022-02-09 by onyeibo

onyeibo, 2022-03-01:
Hi, [About this package](https://download.copr.fedorainfracloud.org/results/sunwire/dkms-rtl8192eu/fedora-rawhide-x86_64/03294042-dkms-rtl8192eu/dkms-rtl8192eu-5.6.4-1.git20220202.fc36.src.rpm)

I recently did a fresh installation of Fedora Rawhide (Server).  The package above installs but the 8192eu module never loads.  lsmod shows the default rtl8xxxu module.  However, ./install_wifi.sh in the git repo does a complete job. I thought you may want to know that the package does not work as expected.

sunwire, 2022-03-01:
> The package above installs but the 8192eu module never loads. lsmod shows the default rtl8xxxu module.

Thanks for the report. I think  rtl8xxxu module is a problem should be blacklisted (./install_wifi.sh script does it)
I made the right changes and now it should work. If you can [test it](https://download.copr.fedorainfracloud.org/results/sunwire/dkms-rtl8192eu/fedora-rawhide-x86_64/03592714-dkms-rtl8192eu/dkms-rtl8192eu-5.6.4-2.git20220202.fc37.noarch.rpm), I would be grateful

onyeibo, 2022-03-08:
>  If you can [test it](https://download.copr.fedorainfracloud.org/results/sunwire/dkms-rtl8192eu/fedora-rawhide-x86_64/03592714-dkms-rtl8192eu/dkms-rtl8192eu-5.6.4-2.git20220202.fc37.noarch.rpm), I would be grateful

Please advise on how I should reverse the installation done by install_wifi.sh.  I will need to get back to a neutral state to test this new package

sunwire, 2022-03-08:
> Please advise on how I should reverse the installation done by install_wifi.sh. I will need to get back to a neutral state to test this new package

If you have a package installed in the previous version
sudo rpm -e dkms-rtl8182eu

sudo rmmod 8192eu
sudo rmmod rtl8xxxu
sudo dkms remove -m rtl8192eu -v 1.0

sudo rm /etc/modprobe.d/blacklist-rtl8xxxu.conf

onyeibo, 2022-03-08:
That worked.

I have also installed the test-package and it went extremely well.   You should probably consider maintaining this package officially for the Fedora community.  How does that sound?

sunwire, 2022-03-09:
> You should probably consider maintaining this package officially for the Fedora community.

Unfortunately, Fedora does not allow to packaging out-of-tree kernel modules https://docs.fedoraproject.org/en-US/packaging-guidelines/what-can-be-packaged/#_no_external_kernel_modules
But I maintain (at least I try) this driver on COPR  https://copr.fedorainfracloud.org/coprs/sunwire/dkms-rtl8192eu/