clnhub/rtl8192eu-linux #50: Not loading on Kernel 5.18.x  (Fedora Rawhide)

onyeibo, 2022-04-05:
The issue is a replica of #45 except that it now happens in the new kernel series (5.18.x) .  The driver compiles and installs but the module does not load.  It loads for the previous series (Kernel 5.17.x) on the same machine.

sunwire, 2022-04-06:
> The driver compiles and installs but the module does not load.

Are you sure it compiles?
```
/var/lib/dkms/8192eu/5.6.4-2.git20220202.fc37/build/os_dep/osdep_service.c: In function ‘isFileReadable’:
/var/lib/dkms/8192eu/5.6.4-2.git20220202.fc37/build/os_dep/osdep_service.c:2200:9: error: unknown type name ‘mm_segment_t’
 2200 |         mm_segment_t oldfs;
      |         ^~~~~~~~~~~~
/var/lib/dkms/8192eu/5.6.4-2.git20220202.fc37/build/os_dep/osdep_service.c: In function ‘retriveFromFile’:
/var/lib/dkms/8192eu/5.6.4-2.git20220202.fc37/build/os_dep/osdep_service.c:2244:9: error: unknown type name ‘mm_segment_t’
 2244 |         mm_segment_t oldfs;
      |         ^~~~~~~~~~~~
/var/lib/dkms/8192eu/5.6.4-2.git20220202.fc37/build/os_dep/osdep_service.c: In function ‘storeToFile’:
/var/lib/dkms/8192eu/5.6.4-2.git20220202.fc37/build/os_dep/osdep_service.c:2286:9: error: unknown type name ‘mm_segment_t’
 2286 |         mm_segment_t oldfs;
      |         ^~~~~~~~~~~~
```

uname -a
`Linux fedora 5.18.0-0.rc1.18.fc37.x86_64 #1 SMP PREEMPT_DYNAMIC Mon Apr 4 14:49:43 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux`

dkms status
`8192eu/5.6.4-2.git20220202.fc37, 5.17.0-0.rc6.109.fc37.x86_64, x86_64: installed`

CGarces, 2022-04-06:
For 5.18 you need something like this
https://github.com/cilynx/rtl88x2bu/commit/1751f39dcbc2ab101cb17a09052945613bb9b78c

sunwire, 2022-04-07:
@onyeibo New version is available you can test it.

dkms status
`8192eu/5.6.4-2.git20220407.fc37, 5.18.0-0.rc1.18.fc37.x86_64, x86_64: installed`

onyeibo, 2022-04-09:
> @onyeibo New version is available you can test it.
> 
> dkms status `8192eu/5.6.4-2.git20220407.fc37, 5.18.0-0.rc1.18.fc37.x86_64, x86_64: installed`

It is installed and running on the latest Kernel (5.18.0-0.rc1.20220408git1831fed559732b1.20.fc37.x86_64)

However, there were some error messages.  I don't know what to make of them.

```
2022-04-09T11:19:03+0100 SUBDEBUG Erase: kernel-devel-5.18.0-0.rc0.20220401gite8b767f5e04097a.15.fc37.x86_64
2022-04-09T11:27:03+0100 INFO dkms: running auto installation service for kernel 5.18.0-0.rc1.20220408git1831fed559732b1.20.fc37.x86_64
Deprecated feature: REMAKE_INITRD
Deprecated feature: REMAKE_INITRD

Building module:
cleaning build area...
'make' KVER=5.18.0-0.rc1.20220408git1831fed559732b1.20.fc37.x86_64...............................................................................
cleaning build area...

8192eu.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.18.0-0.rc1.20220408git1831fed559732b1.20.fc37.x86_64/extra/
depmod..........
Deprecated feature: REMAKE_INITRD

Building module:
cleaning build area...
'make' all KVER=5.18.0-0.rc1.20220408git1831fed559732b1.20.fc37.x86_64.........................(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.18.0-0.rc1.20220408git1831fed559732b1.20.fc37.x86_64 (x86_64)
Consult /var/lib/dkms/rtl8192eu/1.0/build/make.log for more information.
 Done. 
dkms: running auto installation service for kernel 5.18.0-0.rc1.20220408git1831fed559732b1.20.fc37.x86_64
Deprecated feature: REMAKE_INITRD
Deprecated feature: REMAKE_INITRD
Deprecated feature: REMAKE_INITRD

Building module:
cleaning build area...
'make' all KVER=5.18.0-0.rc1.20220408git1831fed559732b1.20.fc37.x86_64.........................(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.18.0-0.rc1.20220408git1831fed559732b1.20.fc37.x86_64 (x86_64)
Consult /var/lib/dkms/rtl8192eu/1.0/build/make.log for more information.
 Done. 
```

sunwire, 2022-04-09:
@onyeibo Did you update to the latest version of the rtl8192eu driver? If you use the driver from [my COPR repository](https://copr.fedorainfracloud.org/coprs/sunwire/dkms-rtl8192eu/), it should be version 5.6.4-2.git20220407.
To have the latest versions, it is best to enable COPR repository and update or install the driver.
e.g.
```
dnf copr enable sunwire/dkms-rtl8192eu
dnf install dkms-rtl8192eu
```
or if you already have an enabled repository and the driver installed
`dnf update dkms-rtl8192eu `

If you installed the driver from the git repository you have to update it as well.
**Do not install both choose the only one. Git or COPR repo**

```
rpm -q dkms-rtl8192eu 
dkms-rtl8192eu-5.6.4-2.git20220407.fc37.noarch
```

```
dkms status
8192eu/5.6.4-2.git20220407.fc37, 5.18.0-0.rc1.18.fc37.x86_64, x86_64: installed
8192eu/5.6.4-2.git20220407.fc37, 5.18.0-0.rc1.20220408git1831fed559732b1.20.fc37.x86_64, x86_64: installed
```

onyeibo, 2022-04-09:
I did all that.

Here is what I get from dkms status:

```
dkms status
8192eu/5.6.4-2.git20220407.fc37, 5.17.0-0.rc8.123.fc37.x86_64, x86_64: installed
8192eu/5.6.4-2.git20220407.fc37, 5.18.0-0.rc1.20220408git1831fed559732b1.20.fc37.x86_64, x86_64: installedDeprecated feature: REMAKE_INITRD

rtl8192eu/1.0, 5.17.0-0.rc8.123.fc37.x86_64, x86_64: installedDeprecated feature: REMAKE_INITRD
 (WARNING! Diff between built and installed module!)
```

sunwire, 2022-04-09:
> 8192eu/5.6.4-2.git20220407.fc37, 5.18.0-0.rc1.20220408git1831fed559732b1.20.fc37.x86_64, x86_64: installedDeprecated feature: REMAKE_INITRD
> rtl8192eu/1.0, 5.17.0-0.rc8.123.fc37.x86_64, x86_64: installedDeprecated feature: REMAKE_INITRD

I think you have two versions. One from COPR 8192eu/5.6.4-2.git20220407.fc37 and the other from git rtl8192eu/1.0.
Choose only one and the other remove.

sunwire, 2022-04-09:
> 8192eu/5.6.4-2.git20220407.fc37, 5.18.0-0.rc1.20220408git1831fed559732b1.20.fc37.x86_64, x86_64: installedDeprecated feature: REMAKE_INITRD

The code is from COPR repo, but is dkms.conf from git? https://github.com/clnhub/rtl8192eu-linux/blob/8396a4ebb4bde6b5c919d291838320f0e5b480dd/dkms.conf#L5

onyeibo, 2022-04-09:
I've fixed it.  I simply removed the offending kernel

```
dkms status
8192eu/5.6.4-2.git20220407.fc37, 5.18.0-0.rc1.20220408git1831fed559732b1.20.fc37.x86_64, x86_64: installed
```
Now we watch to see how this propagates into newer kernels

onyeibo, 2022-04-14:
> Now we watch to see how this propagates into newer kernels

Now running Fedora Rawhide with Kernel-5.18.0-0.rc2.23.fc37.x86_64.  8192eu/5.6.4-2.git20220407.fc37 is installed and running.  I guess this is a wrap.  Thank you @sunwire and @CGarces

I am closing this ticket now

closed 2022-04-14 by onyeibo