clnhub/rtl8192eu-linux #51: Unable to install on Kali 5.15amd64

Xalb, 2022-05-11:
Running ./install_wifi.sh gives me the following error: 

`rmmod: ERROR: Module 8192eu is not currently loaded
rmmod: ERROR: Module rtl8xxxu is not currently loaded
Error! The module rtl8192eu 1.0 is not currently installed.
This module is not currently ACTIVE for kernel 5.15.0-kali3-amd64 (x86_64).
Error! There is no instance of rtl8192eu 1.0 for kernel 5.15.0-kali3-amd64 (x86_64) located in the DKMS tree.
Creating symlink /var/lib/dkms/rtl8192eu/1.0/source -> /usr/src/rtl8192eu-1.0
Error! Your kernel headers for kernel 5.15.0-kali3-amd64 cannot be found at /lib/modules/5.15.0-kali3-amd64/build or /lib/modules/5.15.0-kali3-amd64/source.
You can use the --kernelsourcedir option to tell DKMS where it's located.
modprobe: FATAL: Module 8192eu not found in directory /lib/modules/5.15.0-kali3-amd64
`

bax-by, 2022-09-25:
could you tell me how to solve this problem? I can't ask for help. Thanks

clnhub, 2022-10-16:
Please install kernel headers.

closed 2025-01-02 by clnhub