clnhub/rtl8192eu-linux #53: Is there a solution for change the network without need to reload the driver?

antonypro, 2022-07-08:
Hi @clnhub, i have a TL-WN8200ND and your's driver enable achieve maximum speed of my dongle, however the network change issue is a little inconvenience, even while `sudo rmmod 8192eu && sudo modprobe 8192eu` workaround works.

Have you a suspect for the problem? I'm not experienced with drivers, but as a programmer, i could offer to work together with you to debug this problem.

Thanks!

clnhub, 2022-07-19:
Did you try 5.6.3.1 or 5.6.4?

antonypro, 2022-07-27:
@clnhub: I've tried branch 5.6.4, in fact it solves the re-connection issue, however the internet speed is significantly slow!

Is there a possibility to merge both branches? With high speed and no re-connection issue?

Thanks!

clnhub, 2022-08-01:
And did you try 5.6.3.1?

The re-connection issues was fixed by a kernel compatibility fix a while ago for branches 5.6.3.1 and 5.6.4. The master branch has much older code, so there is probably a larger fix necessary.

antonypro, 2022-08-12:
@clnhub: I've also tried branch 5.6.3.1, and the speed of connection is similar to 5.6.4.

You said that fix the master branch may be hard, so i will wait.

Thanks for your patience.

closed 2022-08-12 by antonypro