clnhub/rtl8192eu-linux #54: Kernel freeze on adding second interface

Zur13, 2022-08-10:
Hi @clnhub I'm trying to use this driver for pentest and when setting up the second interface I got kernel freeze. Can you help me to solve this problem? I have some programming experience but not in C or Linux but I can do my best to provide you the information needed.

To reproduce this issue:
1.  Turn on monitor mode feature in Makefile: `CONFIG_WIFI_MONITOR = y`
2.  Then make and install driver `$ make all` and `# make install`
3.  Disable default driver: `# echo "blacklist rtl8xxxu" >> /etc/modprobe.d/blacklist-rtl8xxxu.conf `
4.  Stop networking services that might mess up the tests `$ sudo systemctl stop NetworkManager` and `$ sudo airmon-ng check kill`
5.  Attach usb dongle
6.  Switch dongle to wifi monitor mode `# ifconfig wlan0 down && iw wlan0 set monitor control && ifconfig wlan0 up`
7.  **Kernel freezes on this step!** Try to add the second network interface to the same dongle `w dev wlan0 interface add wlan0ap type monitor`

Tested on branch 5.6.4 with 
``` 
$ uname -a
Linux kali 5.18.0-kali2-amd64 #1 SMP PREEMPT_DYNAMIC Debian 5.18.5-1kali1 (2022-06-20) x86_64 GNU/Linux
```

<details> 
  <summary>This is what I found in dmesg </summary>

```
[  605.192743] INFO: task vminfo:788 blocked for more than 120 seconds.
[  605.192749]       Tainted: G        W  OE     5.18.0-kali2-amd64 #1 Debian 5.18.5-1kali1
[  605.192750] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  605.192751] task:vminfo          state:D stack:    0 pid:  788 ppid:     1 flags:0x00000000
[  605.192754] Call Trace:
[  605.192755]  <TASK>
[  605.192758]  __schedule+0x30b/0x9e0
[  605.192763]  schedule+0x4e/0xb0
[  605.192764]  schedule_preempt_disabled+0x14/0x20
[  605.192766]  __mutex_lock.constprop.0+0x2b8/0x460
[  605.192767]  dev_ifconf+0xa0/0x150
[  605.192770]  sock_ioctl+0x1c8/0x300
[  605.192773]  __x64_sys_ioctl+0x82/0xb0
[  605.192776]  do_syscall_64+0x3b/0xc0
[  605.192778]  entry_SYSCALL_64_after_hwframe+0x44/0xae
[  605.192780] RIP: 0033:0x7fee3b46d397
[  605.192781] RSP: 002b:00007fee39efab28 EFLAGS: 00000246 ORIG_RAX: 0000000000000010
[  605.192783] RAX: ffffffffffffffda RBX: 0000000000000100 RCX: 00007fee3b46d397
[  605.192784] RDX: 00007fee39efac90 RSI: 0000000000008912 RDI: 0000000000000004
[  605.192784] RBP: 00007fee39eface0 R08: 00007fee30008be0 R09: 00007fee30000080
[  605.192785] R10: 00007fee30008ce0 R11: 0000000000000246 R12: 00007fee39efac90
[  605.192786] R13: 00007fee30008be0 R14: 000000000069b821 R15: 0000000000000004
[  605.192787]  </TASK>
[  605.192799] INFO: task iw:1615 blocked for more than 120 seconds.
[  605.192800]       Tainted: G        W  OE     5.18.0-kali2-amd64 #1 Debian 5.18.5-1kali1
[  605.192801] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  605.192802] task:iw              state:D stack:    0 pid: 1615 ppid:  1372 flags:0x00004000
[  605.192803] Call Trace:
[  605.192804]  <TASK>
[  605.192805]  __schedule+0x30b/0x9e0
[  605.192806]  schedule+0x4e/0xb0
[  605.192807]  schedule_preempt_disabled+0x14/0x20
[  605.192808]  __mutex_lock.constprop.0+0x2b8/0x460
[  605.192811]  cfg80211_netdev_notifier_call+0xf1/0x4e0 [cfg80211]
[  605.192827]  ? __ipv6_dev_mc_inc+0x127/0x380
[  605.192829]  ? netlink_broadcast+0x13a/0x480
[  605.192832]  ? kmem_cache_alloc_trace+0x177/0x2a0
[  605.192833]  ? igmp6_group_added+0xc0/0x100
[  605.192835]  ? __ipv6_dev_mc_inc+0x241/0x380
[  605.192836]  ? ipv6_add_dev+0x2c6/0x510
[  605.192838]  ? addrconf_notify+0x2bf/0x930
[  605.192840]  ? _raw_spin_lock_irqsave+0x24/0x50
[  605.192842]  raw_notifier_call_chain+0x44/0x60
[  605.192856]  register_netdevice+0x491/0x5d0
[  605.192858]  cfg80211_rtw_add_virtual_intf+0x170/0x23e [8192eu]
[  605.192886]  nl80211_new_interface+0x1b5/0x4c0 [cfg80211]
[  605.192901]  genl_family_rcv_msg_doit+0xea/0x150
[  605.192907]  genl_rcv_msg+0xdc/0x1e0
[  605.192908]  ? nl80211_get_interface+0xa0/0xa0 [cfg80211]
[... 239 lines cut ...]
[  726.025431]  rtnetlink_rcv_msg+0xf7/0x380
[  726.025433]  ? dput+0x17b/0x2e0
[  726.025435]  ? rtnl_calcit.isra.0+0x130/0x130
[  726.025436]  netlink_rcv_skb+0x51/0xf0
[  726.025438]  netlink_unicast+0x23a/0x360
[  726.025439]  netlink_sendmsg+0x24f/0x4b0
[  726.025440]  sock_sendmsg+0x5e/0x60
[  726.025441]  __sys_sendto+0xf0/0x160
[  726.025442]  ? __seccomp_filter+0x386/0x590
[  726.025445]  __x64_sys_sendto+0x20/0x30
[  726.025446]  do_syscall_64+0x3b/0xc0
[  726.025447]  entry_SYSCALL_64_after_hwframe+0x44/0xae
[  726.025448] RIP: 0033:0x7fa1293c55c6
[  726.025449] RSP: 002b:00007ffd71649d98 EFLAGS: 00000246 ORIG_RAX: 000000000000002c
[  726.025449] RAX: ffffffffffffffda RBX: 000000001a0e6553 RCX: 00007fa1293c55c6
[  726.025450] RDX: 0000000000000020 RSI: 000055e173bf7f80 RDI: 0000000000000006
[  726.025451] RBP: 000055e173c710f0 R08: 00007ffd71649e10 R09: 0000000000000080
[  726.025451] R10: 0000000000000000 R11: 0000000000000246 R12: 0000000000000001
[  726.025452] R13: 0000000000000000 R14: 0000000000000000 R15: 000055e173bef730
[  726.025453]  </TASK>
```

</details>

Thanks!

clnhub, 2022-08-10:
This is probably a not-implemented freeze.

You can try `iw list` to see the supported (software) modes/interfaces.

Zur13, 2022-08-10:
The monitor mode itself works fine after Makefile changes from step 1 `CONFIG_WIFI_MONITOR = y`.  Command on step 6 successfully switches dongle to wifi monitor mode and I've tested device in this mode and it works as it should. 

The problem is the command on step 7. Before executing step 7 system and device are fully functional. Step 7 is needed to assign the second software interface named wlan0ap to the same hardware  so the single usb dongle would have wlan0 and wlan0ap interfaces at the same time. As I understand from tutorial I follow it should be supported by most chipsets. 

Is it possible to know which function is required and was not implemented for this feature?

clnhub, 2022-08-11:
You could try the default (rtl8xxxu) driver.

My advice however would be to get a wifi chip that mentions support for interface combinations.

Zur13, 2022-08-11:
Yes I've tried default driver rtl8xxxu before and it has a problem with monitor mode (it is possible to turn it on but it does not work at all). 

I've spend quite a lot of time trying to understand what is going on here but unfortunately I have no more time to spend so this is what I've found. As far as I understand the problem is not in the chipset capabilities but in driver code which stops in some mutex forever which might be related to the rtnl_lock or it might be related to this [change](https://github.com/torvalds/linux/commit/2fe8ef106238b274c505c480ecf00d8765abf0d8?diff=split) in the kernel. 

Thanks for your help anyway.

jameshilliard, 2022-10-14:
I wonder if #59 might fix this.

oddsito, 2023-04-16:
happens to me too .

closed 2025-01-02 by clnhub