clnhub/rtl8192eu-linux #55: Not Loading for Kernel-6.0.0-0* Series (Fedora Rawhide)

onyeibo, 2022-09-23:
So far, rtl8192eu-linux compiles and loads for kernel-5.19* series.  However, Fedora is set to release v37 and it appears the community is planning to do that with kernel 6.  Sadly, rtl8192eu-linux does not load for that series.

I have tested loading against several Kernel-6 packages in Fedora.  I am currently on 6.0.0-0.rc6.41.fc38 (not loading)

Please react.

sunwire, 2022-09-23:
Are you sure about this? I don't have a card with this chipset, but it compiles, installs and loads without a problem. 
OS: Fedora Rawhide (38), secure boot disabled
```
[root@fedora ~]# dkms status 
8192eu/5.6.4-2.git20220820.fc38, 6.0.0-0.rc6.41.fc38.x86_64, x86_64: installed
[root@fedora ~]# lsmod |grep 8192eu
8192eu               1519616  0
cfg80211             1081344  1 8192eu
[root@fedora ~]# cat /etc/os-release 
NAME="Fedora Linux"
VERSION="38 (Workstation Edition Prerelease)"
```

onyeibo, 2022-09-23:
Suddenly, I cannot rreplicate my issue.  This is most embarrassing.  I was almost certain it didn't load after the update.  Perhaps it needed a reboot.  In any case, I am sorry for the false alarm.

```
[schemefusion@one ~]$ dkms status
8192eu/5.6.4-2.git20220820.fc38, 5.19.0-65.fc37.x86_64, x86_64: installed
8192eu/5.6.4-2.git20220820.fc38, 6.0.0-0.rc6.41.fc38.x86_64, x86_64: installed
[schemefusion@one ~]$ uname -a
Linux one.xxxxxx.xxx 6.0.0-0.rc6.41.fc38.x86_64 #1 SMP PREEMPT_DYNAMIC Mon Sep 19 14:02:06 UTC 2022 x86_64 GNU/Linux
[schemefusion@one ~]$ lsmod | grep 8192eu
8192eu               1519616  0
cfg80211             1081344  1 8192eu
[schemefusion@one ~]$
```

closed 2022-09-23 by onyeibo