clnhub/rtl8192eu-linux #58: Incorrect netdev->dev_addr on 5.11.2.1 branch

jameshilliard, 2022-10-14:
While testing on the [`5.11.2.1`](https://github.com/clnhub/rtl8192eu-linux/tree/5.11.2.1) branch with kernel `6.0.1` I'm seeing this error in dmesg, although the driver does appear to work regardless.
```c
[  329.010970] ------------[ cut here ]------------
[  329.011527] netdevice: wlp0s5u1: Incorrect netdev->dev_addr
[  329.012189] WARNING: CPU: 2 PID: 488 at net/core/dev_addr_lists.c:519 dev_addr_check.cold+0x43/0x74
[  329.013300] Modules linked in: 8192eu(O) xt_mark mac80211 libarc4 cfg80211 efivarfs [last unloaded: 8192eu(O)]
[  329.014532] CPU: 2 PID: 488 Comm: NetworkManager Tainted: G           O       6.0.1 #1
[  329.015465] Hardware name: QEMU Standard PC (Q35 + ICH9, 2009), BIOS 0.0.0 02/06/2015
[  329.016404] RIP: 0010:dev_addr_check.cold+0x43/0x74
[  329.017024] Code: 01 e8 79 25 fa ff 0f 0b 48 c7 c5 a2 2d 89 94 80 3b 00 75 27 48 c7 c6 ad 2d 89 94 48 89 ea 48 c7 c7 10 a8 8c 94 e8 55 25 fa ff <0f> 0b e9 82 54 d9 ff 48 8b 2c d5 40 19 55 94 eb d4 be 25 00 00 00
[  329.019224] RSP: 0018:ffff9e69c03db7b8 EFLAGS: 00010282
[  329.019886] RAX: 0000000000000000 RBX: ffff8c6750f3f000 RCX: 0000000000000000
[  329.020764] RDX: 0000000000000001 RSI: ffffffff947e62f1 RDI: 00000000ffffffff
[  329.021600] RBP: ffffffff947e62f1 R08: 00000000ffffdfff R09: 00000000ffffdfff
[  329.022450] R10: ffffffff94a59f60 R11: ffffffff94a59f60 R12: ffffffffc05ec400
[  329.023327] R13: ffff8c6750f3f250 R14: 0000000000000000 R15: 0000000000000001
[  329.024212] FS:  00007f86d8a943c0(0000) GS:ffff8c68b7d00000(0000) knlGS:0000000000000000
[  329.025194] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  329.026000] CR2: 00007f2ea1408bd0 CR3: 0000000105b30000 CR4: 00000000003506e0
[  329.026860] Call Trace:
[  329.027174]  <TASK>
[  329.027441]  __dev_open+0x3a/0x1a0
[  329.027868]  __dev_change_flags+0x1a1/0x210
[  329.028357]  dev_change_flags+0x1d/0x60
[  329.028840]  do_setlink+0x7fd/0x12b0
[  329.029285]  ? __nla_validate_parse+0x5f/0xc40
[  329.029853]  __rtnl_newlink+0x618/0x9b0
[  329.030326]  rtnl_newlink+0x3e/0x60
[  329.030747]  rtnetlink_rcv_msg+0x149/0x3b0
[  329.031235]  ? rtnl_calcit.isra.0+0x130/0x130
[  329.031784]  netlink_rcv_skb+0x4c/0xf0
[  329.032259]  netlink_unicast+0x23d/0x390
[  329.032764]  netlink_sendmsg+0x23b/0x4b0
[  329.033242]  sock_sendmsg+0x5a/0x60
[  329.033678]  ____sys_sendmsg+0x227/0x260
[  329.034168]  ? copy_msghdr_from_user+0x78/0xb0
[  329.034749]  ___sys_sendmsg+0x95/0xe0
[  329.035224]  __sys_sendmsg+0x71/0xc0
[  329.035685]  do_syscall_64+0x3a/0x90
[  329.036152]  entry_SYSCALL_64_after_hwframe+0x63/0xcd
[  329.036809] RIP: 0033:0x7f86db53eb32
[  329.037255] Code: 28 89 54 24 1c 48 89 74 24 10 89 7c 24 08 e8 f9 9e f9 ff 8b 54 24 1c 48 8b 74 24 10 41 89 c0 8b 7c 24 08 b8 2e 00 00 00 0f 05 <48> 3d 00 f0 ff ff 76 10 48 8b 15 b7 d2 0a 00 f7 d8 64 89 02 48 83
[  329.039445] RSP: 002b:00007fff332741a0 EFLAGS: 00000293 ORIG_RAX: 000000000000002e
[  329.040368] RAX: ffffffffffffffda RBX: 0000564ea7a2ad00 RCX: 00007f86db53eb32
[  329.041256] RDX: 0000000000000000 RSI: 00007fff332741f0 RDI: 000000000000000c
[  329.042129] RBP: 00007fff332741f0 R08: 0000000000000000 R09: 0000000000000000
[  329.043008] R10: 0000000000000010 R11: 0000000000000293 R12: 0000000000000000
[  329.043875] R13: 00007fff33274398 R14: 00007fff3327438c R15: 0000000000000000
[  329.044914]  </TASK>
[  329.045327] ---[ end trace 0000000000000000 ]---
```

jameshilliard, 2022-10-14:
> Are you on Ubuntu 22.10?

Nope, I'm using [buildroot](https://buildroot.org/).

> I've tested the driver on 22.04 with 6.0.1 and there were are no (x86) issues.

I'm running a [stock 6.0.1 kernel](https://cdn.kernel.org/pub/linux/kernel/v6.x/linux-6.0.1.tar.xz).

> This seems to be a bug in 22.10 unrelated to the driver.

It def gets triggered by modprobing the 8192eu driver.

> This looks similar: [https://bugs.launchpad.net/ubuntu/+source/broadcom-sta/+bug/1992269](https://bugs.launchpad.net/ubuntu/+source/broadcom-sta/+bug/1992269)

It does indeed, however since that's reported against a different wifi driver(one for broadcom hardware) which I'm not using at all here I suspect that something got changed in newer kernel versions which may have triggered an issue in both drivers.

clnhub, 2022-10-14:
Do you have the same trace with an older branch?

jameshilliard, 2022-10-14:
> Do you have the same trace with an older branch?

Haven't tried but they probably have the same bug, should be fixed by #59 which probably should be backported to older branches.

clnhub, 2022-10-14:
Can you rename to "Improve support for kernel 5.15(+)", this makes it easier to find later on.

Also, did you verify that this works?

jameshilliard, 2022-10-14:
> Can you rename to "Improve support for kernel 5.15(+)", this makes it easier to find later on.

Updated PR to that.

> Also, did you verify that this works?

Yeah, from what I can tell it fully fixes the issue.

PR #59 (2022-10-14): Improve support for kernel 5.15(+)
Fixes #58.
closed 2022-10-16 by clnhub