clnhub/rtl8192eu-linux #60: 8192eu: exec format error on Raspberry Pi 2

suryansh011, 2022-10-25:
I'm trying to install the module on "SARPi Project Slackware 15.0 ARM"

`root@nemesis:~# uname -a
Linux nemesis.nemesis.lan 5.15.73-v7-sarpi2 #1 SMP Sat Oct 15 09:08:55 BST 2022 armv7l BCM2835 GNU/Linux
`

`root@nemesis:~# modprobe 8192eu
modprobe: ERROR: could not insert '8192eu': Exec format error
`

`root@nemesis:/lib/modules/5.15.73-v7-sarpi2/kernel/drivers/net/wireless# insmod 8192eu.ko.xz
insmod: ERROR: could not insert module 8192eu.ko.xz: Invalid module format
root@nemesis:/lib/modules/5.15.73-v7-sarpi2/kernel/drivers/net/wireless#
`

I couldn't make it work even after hours of effort.

clnhub, 2022-10-25:
You probably need to alter the MakeFile. Please try the `CONFIG_PLATFORM_ARM_RPI` preset.

Add:

`CONFIG_PLATFORM_ARM_RPI = y`

and set 
`
CONFIG_PLATFORM_I386_PC = n`

Add:
```
ifeq ($(CONFIG_PLATFORM_ARM_RPI), y)
EXTRA_CFLAGS += -DCONFIG_LITTLE_ENDIAN
EXTRA_CFLAGS += -DCONFIG_IOCTL_CFG80211 -DRTW_USE_CFG80211_STA_EVENT
ARCH := arm
CROSS_COMPILE :=
KVER  := $(shell uname -r)
KSRC ?= /lib/modules/$(KVER)/build
MODDESTDIR := /lib/modules/$(KVER)/kernel/drivers/net/wireless/
endif
```

suryansh011, 2022-10-26:
> You probably need to alter the MakeFile. Please try the `CONFIG_PLATFORM_ARM_RPI` preset.
>
> Add:
>
> `CONFIG_PLATFORM_ARM_RPI = y`
>
> and set ` CONFIG_PLATFORM_I386_PC = n`
>
> Add:
>
> ```
> ifeq [...]

It still doesn't work.

`
root@nemesis:~/rtl8192eu-linux# ./install_wifi.sh
rmmod: ERROR: Module 8192eu is not currently loaded
rmmod: ERROR: Module rtl8xxxu is not currently loaded
Deprecated feature: REMAKE_INITRD
Module rtl8192eu-1.0 for kernel 5.15.73-v7-sarpi2 (armv7l).
Before uninstall, this module version was ACTIVE on this kernel.

8192eu.ko.xz:
 - Uninstallation
   - Deleting from: /lib/modules/5.15.73-v7-sarpi2/kernel/drivers/net/wireless//
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.
depmod.........
Deprecated feature: REMAKE_INITRD
Module rtl8192eu 1.0 is not installed for kernel 5.15.73-v7-sarpi2 (armv7l). Skipping...
Deleting module rtl8192eu-1.0 completely from the DKMS tree.
rm: cannot remove '/var/lib/dkms/rtl8192eu': No such file or directory
Deprecated feature: REMAKE_INITRD
Creating symlink /var/lib/dkms/rtl8192eu/1.0/source -> /usr/src/rtl8192eu-1.0
Sign command: /lib/modules/5.15.73-v7-sarpi2/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Binary sign-file not found, module won't be signed
Deprecated feature: REMAKE_INITRD

Building module:
Cleaning build area...
'make' all KVER=5.15.73-v7-sarpi2................................................................................................................................................................................................................................................................................................................................................................................................................................................
Cleaning build area...

8192eu.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.15.73-v7-sarpi2/kernel/drivers/net/wireless//
depmod.........
modprobe: ERROR: could not insert '8192eu': Exec format error
`

clnhub, 2022-10-26:
Is you current (module) GCC version (`gcc --version`) the same as the one used to build the kernel (`cat /proc/version`)?

Kernel headers and the kernel, same version?

Which version of the driver did you try?

suryansh011, 2022-10-26:
> Is you current (module) GCC version (`gcc --version`) the same as the one used to build the kernel (`cat /proc/version`)?
> 
> Kernel headers and the kernel, same version?
> 
> Which version of the driver did you try?

It works! My bad, it was an issue with the bad kernel source.
Now it works perfectly!

Thank you so much for helping out! :))

closed 2022-10-26 by suryansh011

A-lexLuthor, 2023-09-30:
> It works! My bad, it was an issue with the bad kernel source. Now it works perfectly!
>
> Thank you so much for helping out! :))

can you explain how you resolve this issuse ?