clnhub/rtl8192eu-linux #64: driver apparently not working on fedoraOS 36 kernel 6.0.8-200.fc36.x86_64

bdpor, 2022-11-15:
This is my adaptor's ID from _lsusb_: 
Bus 001 Device 006: ID 2001:3319 D-Link Corp. DWA-131 Wireless N Nano Adapter (Rev. E1) [Realtek RTL8192EU]

The current distro is Fedora Linux 36 (Workstation Edition) on a Surface Pro 3 hardware. (of note the Marvell wifi card of the SP3 is disable on UEFI settings, as the card has not worked on windows for years now).
The current kernel: 6.0.8-200.fc36.x86_64

To install the driver I have downloaded it from here, namely from the following branch: rtl8192eu-linux-5.11.2.1
Followed the manual instructions, because the automatic installer did not work, it reported an error. I am sorry but I do not have the output of the error available. However, manual installation completed without any error message.

rtl8xxxu was blacklisted

Upon _sudo dkms status_ the following output is obtained:
rtl8192eu/1.0, 6.0.7-200.fc36.x86_64, x86_64: installed
rtl8192eu/1.0, 6.0.8-200.fc36.x86_64, x86_64: installed

However, when I plug the adaptor in and confirm that it is connected with _lsusb_ no wifi network is available on the settings app. 
Nevertheless, before blacklisting rtl8xxxu or installing the rtl8192eu driver I was able to view available networks albeit without completing a successful connection.

I have also tried it with TL-W823N adaptor with the same results described above.

Is there anything I am doing wrong or could do better?
I am trying to get linux working on the go in this SP3, and been unsuccessful thus far because I cannot get a working wifi (with a USB dongle) and only USB tethereing to LAN or mobile phone as the only available option for network access.

sunwire, 2022-11-15:
Did you blacklist rtl8xxxu module?
Is the module loaded? ( lsmod|grep 8192eu )

bdpor, 2022-11-15:
@sunwire , thank you for the feedback

> Did you blacklist rtl8xxxu module? 
Yes, as far as I know how.

$ ls [directory /etc/modprobe.d/]
appletalk-blacklist.conf   kvm.conf                     nfc-blacklist.conf
atm-blacklist.conf         l2tp_eth-blacklist.conf      nvdimm-security.conf
ax25-blacklist.conf        l2tp_ip-blacklist.conf       rds-blacklist.conf
batman-adv-blacklist.conf  l2tp_netlink-blacklist.conf  rose-blacklist.conf
blacklist-rtl8xxxu.conf    l2tp_ppp-blacklist.conf      rtl8xxxu.conf
firewalld-sysctls.conf     lockd.conf                   sctp-blacklist.conf
floppy-blacklist.conf      netrom-blacklist.conf        vhost.conf

  Both: _blacklist-rtl8xxxu.conf_    and     _rtl8xxxu.conf _ files when opened in nano have but one sentence "blacklist rtl8xxxu"

> Is the module loaded? ( lsmod|grep 8192eu )
I have rebooted several times since completing the install with no success.
Furthermore, even after running the command _lsmod|grep 8192eu_ there is still no available wifi network or any recognition of a wifi card in the definitions app.

clnhub, 2022-11-16:
Could you add the output of the installer and `sudo dmesg | grep 8192eu` when booting with the USB-stick inserted?

bdpor, 2022-11-16:
> 

The system has a dual boot with windows 10. Booted to Windows 10 with the USB and the USB drive is working.
Rebooted the system into fedoraOS 36, with the kernel 6.0.8-200.fc36.x86_64 without removing the USB dongle.

Opened terminal:

`$ lsusb`
Bus 002 Device 002: ID 0bda:0307 Realtek Semiconductor Corp. Card Reader
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 045e:07bf Microsoft Corp. Microsoft LifeCam Rear
Bus 001 Device 004: ID 045e:07be Microsoft Corp. Microsoft LifeCam Front
Bus 001 Device 003: ID 045e:07dc Microsoft Corp. Surface Type Cover
Bus 001 Device 002: ID 2001:3319 D-Link Corp. DWA-131 Wireless N Nano Adapter (Rev. E1) [Realtek RTL8192EU]
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

`$ sudo dmesg | grep 8192eu`
(no output produced)

Since no output was produced ran:

`$ sudo dkms status`
rtl8192eu/1.0, 6.0.7-200.fc36.x86_64, x86_64: installed
rtl8192eu/1.0, 6.0.8-200.fc36.x86_64, x86_64: installed

Ran the installer:
`$ sudo ./install_wifi.sh`
rmmod: ERROR: Module 8192eu is not currently loaded
rmmod: ERROR: Module rtl8xxxu is not currently loaded
Module rtl8192eu-1.0 for kernel 6.0.8-200.fc36.x86_64 (x86_64).
Before uninstall, this module version was ACTIVE on this kernel.

8192eu.ko.xz:
 - Uninstallation
   - Deleting from: /lib/modules/6.0.8-200.fc36.x86_64/extra/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.
depmod.......
Module rtl8192eu 1.0 is not installed for kernel 6.0.8-200.fc36.x86_64 (x86_64). Skipping...
Creating symlink /var/lib/dkms/rtl8192eu/1.0/source -> /usr/src/rtl8192eu-1.0
Sign command: /lib/modules/6.0.8-200.fc36.x86_64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Certificate or key are missing, generating self signed certificate for MOK...

Building module:
Cleaning build area...
'make' -j4 all KVER=6.0.8-200.fc36.x86_64...........................
Signing module /var/lib/dkms/rtl8192eu/1.0/build/8192eu.ko
At main.c:167:
- SSL error:FFFFFFFF80000002:system library::No such file or directory: crypto/bio/bss_file.c:67
- SSL error:10000080:BIO routines::no such file: crypto/bio/bss_file.c:75
sign-file: /var/lib/dkms/mok.key
Cleaning build area...

8192eu.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.0.8-200.fc36.x86_64/extra/
depmod......
modprobe: ERROR: could not insert '8192eu': Key was rejected by service

If I recall correctly, these are the same errors I ran into the first time I tried to install. After the installer completed the script I ran 

`$ sudo dkms status`
rtl8192eu/1.0, 6.0.8-200.fc36.x86_64, x86_64: installed

Thank you for the support.

bdpor, 2022-11-16:
While the YT video did not load, I followed the text instructions available in the comments.

verified mokutil was installed 
$ `man mokutil`

installed openssl
$ `sudo dnf install openssl`

and then
$ `mokutil --import /var/lib/dkms/mok.pub`
Failed to get file status, /var/lib/dkms/mok.pub

also tried 
$ `sudo mokutil --import /var/lib/dkms/mok.pub`
Failed to get file status, /var/lib/dkms/mok.pub

accessed  /var/lib/dkms/ `ls` listed only rtl8192eu

sunwire, 2022-11-17:
> $ `sudo mokutil --import /var/lib/dkms/mok.pub` Failed to get file status, /var/lib/dkms/mok.pub
> 
> accessed /var/lib/dkms/ `ls` listed only rtl8192eu

Ok, It will be easiest if you remove and install again the driver and when run sudo mokutil --import /var/lib/dkms/mok.pub and reboot

bdpor, 2022-11-18:
@sunwire 

I am sorry to only provide feedback today, but have been unable to test your suggestion earlier.
On your advice I have ran `sudo ./install_wifi.sh` and afterwards the `sudo mokutil --import /var/lib/dkms/mok.pub`
Upon reboot followed the instructions on your YT video and it worked perfectly
This current post is already being written through an internet connection provided from a wifi access that uses the driver provided

Thank you for providing the support neither realtek nor, in my case, D-Link have provided.

closed 2022-11-19 by bdpor