clnhub/rtl8192eu-linux #68: Not working on archlinux with kernel 6.1.1

LamFTS, 2022-12-27:
Dear developer, my device is Mecury MW300UM, I am sure it is embedded with rtl8192eu, I install the driver on my Archlinux with kernel 6.1.1-arch1-1, the driver was build and installed without error, dkms status showed like that:
```
nvidia/390.157, 6.1.1-arch1-1, x86_64: installed (WARNING! Diff between built and installed module!)
rtl8192eu/1.0, 6.1.1-arch1-1, x86_64: installed (original_module exists)
v4l2loopback/0.12.7, 6.1.1-arch1-1, x86_64: installed (WARNING! Diff between built and installed module!)
```
and 8xxxu has been blocked, `ls /etc/modprobe.d/blacklist-rtl8xxxu.conf ` show that:
```
/etc/modprobe.d/blacklist-rtl8xxxu.conf
```, but the device just not working, `lsusb` seems indicated that the system still can't recognized it properly, showing:
```
Bus 002 Device 002: ID 8087:8000 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:8008 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 002: ID 0781:5588 SanDisk Corp. Extreme Pro
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 006: ID 0bda:a192 Realtek Semiconductor Corp. DISK
Bus 003 Device 007: ID 22d9:276a OPPO Electronics Corp. OnePlus 8T
Bus 003 Device 002: ID 413c:2107 Dell Computer Corp. KB212-B Quiet Key Keyboard
Bus 003 Device 005: ID 046d:c077 Logitech, Inc. Mouse
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
, and there is something intriguing that the device stop the system startup proper, when the device is plugged, Archlinux stuck at system-udev, only after remove the device, can the system continue.

clnhub, 2022-12-27:
Did it work with a 5.x/6.0.x kernel?

LamFTS, 2022-12-27:
I switch to the linux-lts kernel which is 5.15.85, the dkms driver was re-complier properly, but the device still not working. I don't know what is going wrong :( .

clnhub, 2022-12-27:
You seem to be having a 0bda:a192 Realtek DISK???? device connected to you computer? Does this disappear after disconnecting the wifi usb stick?

LamFTS, 2022-12-28:
The Realtek Disk disappear after pulling off the device. When I plug it on Windows, there will be a disk containing the driver of this device on my file explorer, so I think this explain why Archlinux recognize it as a DISK, but the disk containing driver didn't show up on my Archlinux, that is strange.

clnhub, 2022-12-28:
You could try switching to wifi mode: `sudo usb_modeswitch -KW -v 0bda -p a192`

LamFTS, 2022-12-28:
it works, oh my god, that is amazing, I got my USB WLAN Adapter work on my Archlinux, thanks a lot. By the way, after mode switch using `usb_modeswitch`, I search the chip model with the new ID, it turn out that my device has RTL8192FU embedded rather than RTL8192EU, after switching to the right mode and installing the right driver, I finally got my device worked, thank for your kindly help again.

closed 2022-12-28 by LamFTS