clnhub/rtl8192eu-linux #70: Driver seems to be broken after recent Kernel update

karlwil, 2023-01-26:
Hey!
Firstly, thanks for making this, you have made installing this driver for Fedora Linux an absolute pleasure.
Now on to the issue...
After a recent Kernel update the installer script breaks:
[user@host rtl8192eu-linux-master]$ sudo ./install_wifi.sh 
rmmod: ERROR: Module 8192eu is not currently loaded
rmmod: ERROR: Module rtl8xxxu is not currently loaded
Module rtl8192eu 1.0 is not installed for kernel 6.1.7-200.fc37.x86_64 (x86_64). Skipping...
Module rtl8192eu 1.0 is not installed for kernel 6.1.7-200.fc37.x86_64 (x86_64). Skipping...
Module rtl8192eu 1.0 is not built for kernel 6.1.7-200.fc37.x86_64 (x86_64). Skipping...
Deleting module rtl8192eu-1.0 completely from the DKMS tree.
rm: cannot remove '/var/lib/dkms/rtl8192eu': No such file or directory
Deprecated feature: REMAKE_INITRD (/usr/src/rtl8192eu-1.0/dkms.conf)
Creating symlink /var/lib/dkms/rtl8192eu/1.0/source -> /usr/src/rtl8192eu-1.0
Sign command: /lib/modules/6.1.7-200.fc37.x86_64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/rtl8192eu/1.0/source/dkms.conf)

Building module:
Cleaning build area...
'make' all KVER=6.1.7-200.fc37.x86_64.................(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.1.7-200.fc37.x86_64 (x86_64)
Consult /var/lib/dkms/rtl8192eu/1.0/build/make.log for more information.
modprobe: FATAL: Module 8192eu not found in directory /lib/modules/6.1.7-200.fc37.x86_64

Taking a look at /var/lib/dkms/rtl8192eu/1.0/build/make.log:
KMS make.log for rtl8192eu-1.0 for kernel 6.1.7-200.fc37.x86_64 (x86_64)
Thu Jan 26 08:36:38 PM GMT 2023
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.1.7-200.fc37.x86_64/build M=/var/lib/dkms/rtl8192eu/1.0/build  modules
make[1]: Entering directory '/usr/src/kernels/6.1.7-200.fc37.x86_64'
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_mlme_ext.o
/var/lib/dkms/rtl8192eu/1.0/build/core/rtw_mlme_ext.c: In function ‘mgt_dispatcher’:
/var/lib/dkms/rtl8192eu/1.0/build/core/rtw_mlme_ext.c:1638:20: warning: this statement may fall through [-Wimplicit-fallthrough=]
 1638 |                 if (MLME_IS_AP(padapter))
      |                    ^
/var/lib/dkms/rtl8192eu/1.0/build/core/rtw_mlme_ext.c:1643:9: note: here
 1643 |         case WIFI_ASSOCREQ:
      |         ^~~~
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_sta_mgt.o
/var/lib/dkms/rtl8192eu/1.0/build/core/rtw_sta_mgt.c: In function ‘rtw_mfree_stainfo’:
/var/lib/dkms/rtl8192eu/1.0/build/core/rtw_sta_mgt.c:312:25: warning: the comparison will always evaluate as ‘true’ for the address of ‘lock’ will never be NULL [-Waddress]
  312 |         if (&psta->lock != NULL)
      |                         ^~
In file included from /var/lib/dkms/rtl8192eu/1.0/build/include/drv_types.h:106,
                 from /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_sta_mgt.c:17:
/var/lib/dkms/rtl8192eu/1.0/build/include/sta_info.h:215:17: note: ‘lock’ declared here
  215 |         _lock   lock;
      |                 ^~~~
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_rson.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_btcoex.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_beamforming.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_odm.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/efuse/rtw_efuse.o
/var/lib/dkms/rtl8192eu/1.0/build/core/efuse/rtw_efuse.c: In function ‘rtw_efuse_analyze’:
/var/lib/dkms/rtl8192eu/1.0/build/core/efuse/rtw_efuse.c:714:17: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
  714 |                 if (i % 16 == 0)
      |                 ^~
In file included from /var/lib/dkms/rtl8192eu/1.0/build/include/drv_types.h:60,
                 from /var/lib/dkms/rtl8192eu/1.0/build/core/efuse/rtw_efuse.c:17:
/var/lib/dkms/rtl8192eu/1.0/build/include/rtw_debug.h:292:9: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘if’
  292 |         do {\
      |         ^~
/var/lib/dkms/rtl8192eu/1.0/build/core/efuse/rtw_efuse.c:716:25: note: in expansion of macro ‘_RTW_PRINT_SEL’
  716 |                         _RTW_PRINT_SEL(RTW_DBGDUMP, "%02X%s"
      |                         ^~~~~~~~~~~~~~
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/os_dep/osdep_service.o
/var/lib/dkms/rtl8192eu/1.0/build/os_dep/osdep_service.c: In function ‘rtw_random32’:
/var/lib/dkms/rtl8192eu/1.0/build/os_dep/osdep_service.c:2478:16: error: implicit declaration of function ‘prandom_u32’; did you mean ‘prandom_u32_max’? [-Werror=implicit-functi>
 2478 |         return prandom_u32();
      |                ^~~~~~~~~~~
      |                prandom_u32_max
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:250: /var/lib/dkms/rtl8192eu/1.0/build/os_dep/osdep_service.o] Error 1
make[1]: *** [Makefile:1996: /var/lib/dkms/rtl8192eu/1.0/build] Error 2
make[1]: Leaving directory '/usr/src/kernels/6.1.7-200.fc37.x86_64'
make: *** [Makefile:1840: modules] Error 2

Any help in fixing this would be greatly appreciated.
Kind regards
Karl

clnhub, 2023-01-27:
I cannot reproduce this on Ubuntu 22.10 with kernel 6.1.8.

Could you try the most recent version of the driver? The _prandom_ issue was fixed a while ago and I can see from the log that you are using an old version of the installer.

sunwire, 2023-01-27:
> Deprecated feature: REMAKE_INITRD (/usr/src/rtl8192eu-1.0/dkms.conf)

I think you use old code, for example this feature has been removed https://github.com/clnhub/rtl8192eu-linux/commit/16ffeecb69a6d97949745413c254234e90ed5923

karlwil, 2023-01-27:
> I cannot reproduce this on Ubuntu 22.10 with kernel 6.1.8.
> 
> Could you try the most recent version of the driver? The _prandom_ issue was fixed a while ago and I can see from the log that you are using an old version of the installer.

Hello clnhub, I am using Fedora, I am going to re-clone and try again, I could very well be using old code. 
I'll update you in a few minutes.

> > Deprecated feature: REMAKE_INITRD (/usr/src/rtl8192eu-1.0/dkms.conf)
> 
> I think you use old code, for example this feature has been removed [16ffeec](https://github.com/clnhub/rtl8192eu-linux/commit/16ffeecb69a6d97949745413c254234e90ed5923) PS. I've noticed you are using Fedora 37 maybe [this](https://copr.fedorainfracloud.org/coprs/sunwire/dkms-rtl8192eu/) helps you .

Thanks sunwire, I'll check this out if I still have issues after re-clone.

Kind regards
Karl

karlwil, 2023-01-27:
Hey @clnhub 
That has fixed the issue, I feel like a chop for not trying that first. Full output below:

`[user@host Downloads]$ git clone https://github.com/clnhub/rtl8192eu-linux
Cloning into 'rtl8192eu-linux'...
remote: Enumerating objects: 2174, done.
remote: Counting objects: 100% (268/268), done.
remote: Compressing objects: 100% (94/94), done.
remote: Total 2174 (delta 240), reused 177 (delta 174), pack-reused 1906
Receiving objects: 100% (2174/2174), 4.13 MiB | 12.00 KiB/s, done.
Resolving deltas: 100% (1383/1383), done.
[user@host Downloads]$ cd rtl8192eu-linux/
[user@host rtl8192eu-linux]$ sudo ./install_wifi.sh 
[sudo] password for cheryl: 
rmmod: ERROR: Module 8192eu is not currently loaded
rmmod: ERROR: Module rtl8xxxu is not currently loaded
Module rtl8192eu 1.0 is not installed for kernel 6.1.7-200.fc37.x86_64 (x86_64). Skipping...
Module rtl8192eu 1.0 is not installed for kernel 6.1.7-200.fc37.x86_64 (x86_64). Skipping...
Module rtl8192eu 1.0 is not built for kernel 6.1.7-200.fc37.x86_64 (x86_64). Skipping...
Deleting module rtl8192eu-1.0 completely from the DKMS tree.
rm: cannot remove '/var/lib/dkms/rtl8192eu': No such file or directory
Creating symlink /var/lib/dkms/rtl8192eu/1.0/source -> /usr/src/rtl8192eu-1.0
Sign command: /lib/modules/6.1.7-200.fc37.x86_64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
'make' -j4 all KVER=6.1.7-200.fc37.x86_64......................................
Signing module /var/lib/dkms/rtl8192eu/1.0/build/8192eu.ko
Cleaning build area...

8192eu.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.1.7-200.fc37.x86_64/extra/
depmod.......
[user@host rtl8192eu-linux]$`

Thanks for the help
Kind regards
Karl

closed 2023-01-27 by karlwil