clnhub/rtl8192eu-linux #71: Unable to Intall on Red Hat ( Rocky Linux 9.1)

enzonun, 2023-02-09:
OBS: I already installed the development tools (dkms, kernel-devel, kernel-headers,etc )

Error description: 

'make' -j6 all KVER=5.14.0-162.12.1.el9_1.0.2.x86_64.......(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.14.0-162.12.1.el9_1.0.2.x86_64 (x86_64)
Consult /var/lib/dkms/rtl8192eu/1.0/build/make.log for more information.
modprobe: FATAL: Module 8192eu not found in directory /lib/modules/5.14.0-162.12.1.el9_1.0.2.x86_64

sunwire, 2023-02-10:
```
/var/lib/dkms/8192eu/5.11.2.1-3.git20221229.fc38/build/os_dep/linux/rtw_proc.c: In function ‘rtw_drv_proc_open’:
/var/lib/dkms/8192eu/5.11.2.1-3.git20221229.fc38/build/os_dep/linux/rtw_proc.c:243:34: error: implicit declaration of function ‘PDE_DATA’; did you mean ‘NODE_DATA’? [-Werror=implicit-function-declaration]
  243 |         ssize_t index = (ssize_t)PDE_DATA(inode);
      |                                  ^~~~~~~~
      |                                  NODE_DATA
```
PDE_DATA() was replaced by pde_data() in kernel 5.17 but it looks that RHEL kernel 5.14 backported some patches

sunwire, 2023-02-11:
@enzonun I you wish, you can test the patched driver from my COPR repo https://copr.fedorainfracloud.org/coprs/sunwire/dkms-rtl8192eu/
You have to install the EPEL repo first. I've test it on AlmaLinux and it compiles and installs without a problem (I have not tested 'cos I don't have compatible hardware).
@clnhub The patch is only applied in the rpm spec file in compatible RHEL systems
[0001-Add-RHEL9.1-and-compatible-kernel-support.txt](https://github.com/clnhub/rtl8192eu-linux/files/10714354/0001-Add-RHEL9.1-and-compatible-kernel-support.txt)

enzonun, 2023-02-11:
> @enzonun I you wish, you can test the patched driver from my COPR repo https://copr.fedorainfracloud.org/coprs/sunwire/dkms-rtl8192eu/ You have to install the EPEL repo first. I've test it on [...]

Thanks! Yes that was the error I find in the code.. I already installed Fedora 37 because of job reasons, but I'll give it a try when I can. Very kind

clnhub, 2023-02-11:
Please try clnhub_test9

sunwire, 2023-02-12:
> Please try clnhub_test9

It compiles and I was able to insert module on both systems Fedora (6.1.10) and AlmaLinux (5.14.0)

clnhub, 2023-02-13:
Thanks @enzonun and @sunwire for the feedback.

closed 2023-02-13 by clnhub