clnhub/rtl8192eu-linux #72: not make in CentOS 9

sneganov, 2023-02-27:
uname -a
Linux centos 5.14.0-274.el9.x86_64 #1 SMP PREEMPT_DYNAMIC Thu Feb 16 14:51:25 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

  CC [M]  /data/drv/9/rtl8192eu-linux/core/efuse/rtw_efuse.o
/data/drv/9/rtl8192eu-linux/core/efuse/rtw_efuse.c: В функции «rtw_efuse_analyze»:
/data/drv/9/rtl8192eu-linux/core/efuse/rtw_efuse.c:714:17: предупреждение: эта «if» клаузула не защищает... [-Wmisleading-indentation]
  714 |                 if (i % 16 == 0)
      |                 ^~
In file included from /data/drv/9/rtl8192eu-linux/include/drv_types.h:60,
                 from /data/drv/9/rtl8192eu-linux/core/efuse/rtw_efuse.c:17:
/data/drv/9/rtl8192eu-linux/include/rtw_debug.h:292:9: замечание: ...этот оператор, но последний ошибочно полагается на защиту «if»
  292 |         do {\
      |         ^~
/data/drv/9/rtl8192eu-linux/core/efuse/rtw_efuse.c:716:25: замечание: в расширении макроса «_RTW_PRINT_SEL»
  716 |                         _RTW_PRINT_SEL(RTW_DBGDUMP, "%02X%s"
      |                         ^~~~~~~~~~~~~~
  CC [M]  /data/drv/9/rtl8192eu-linux/os_dep/osdep_service.o
/data/drv/9/rtl8192eu-linux/os_dep/osdep_service.c: В функции «thread_exit»:
/data/drv/9/rtl8192eu-linux/os_dep/osdep_service.c:1180:9: ошибка: неявная декларация функции «complete_and_exit» [-Werror=implicit-function-declaration]
 1180 |         complete_and_exit(comp, 0);
      |         ^~~~~~~~~~~~~~~~~
/data/drv/9/rtl8192eu-linux/os_dep/osdep_service.c: В функции «isFileReadable»:
/data/drv/9/rtl8192eu-linux/os_dep/osdep_service.c:2057:9: ошибка: неизвестное имя типа «mm_segment_t»
 2057 |         mm_segment_t oldfs;
      |         ^~~~~~~~~~~~
/data/drv/9/rtl8192eu-linux/os_dep/osdep_service.c: В функции «retriveFromFile»:
/data/drv/9/rtl8192eu-linux/os_dep/osdep_service.c:2099:9: ошибка: неизвестное имя типа «mm_segment_t»
 2099 |         mm_segment_t oldfs;
      |         ^~~~~~~~~~~~
/data/drv/9/rtl8192eu-linux/os_dep/osdep_service.c: В функции «storeToFile»:
/data/drv/9/rtl8192eu-linux/os_dep/osdep_service.c:2139:9: ошибка: неизвестное имя типа «mm_segment_t»
 2139 |         mm_segment_t oldfs;
      |         ^~~~~~~~~~~~
cc1: некоторые предупреждения считаются ошибками
make[2]: *** [scripts/Makefile.build:321: /data/drv/9/rtl8192eu-linux/os_dep/osdep_service.o] Ошибка 1
make[1]: *** [Makefile:1923: /data/drv/9/rtl8192eu-linux] Ошибка 2
make[1]: выход из каталога «/usr/src/kernels/5.14.0-274.el9.x86_64»
make: *** [Makefile:1841: modules] Ошибка 2

clnhub, 2023-02-28:
Are you using the latest git code? Please try changing (osdep_service.c, line 1177) `KERNEL_VERSION(5, 17, 0))` to `KERNEL_VERSION(5, 14, 0))`.

If that does not work, you can try the (default) 5.11.2.1 branch. This may require changing the same code @ line 1312.

Updating the kernel to at least 5.17 would probably be the easiest fix.

clnhub, 2023-03-01:
Thanks @sunwire

closed 2023-03-01 by clnhub