clnhub/rtl8192eu-linux #77: Driver no longer loading as from Kernel 6.3 (Fedora Linux)

onyeibo, 2023-03-27:
I have watched this for a long time now and it is consistent.  I reserved the last release of kernel 6.2 on my Fedora box (rawhide version).  I observe that the WIFI driver does not load when I chose any of the recent kernels before booting.  I am stuck with that last release of 6,2.

I hope I am not the only one seeing this.

sunwire, 2023-03-27:
> 5.11.2.1-4.git20230212

Your driver is also a month old (current version 5.11.2.1-5.git20230312 and it include support for kernel 6.3 which was added three weeks ago)

onyeibo, 2023-03-27:
```
Running scriptlet: kernel-modules-core-6.3.0-0.rc3.20230324git1e760fa3596e.33.fc39.x86_64                                                                              
Running scriptlet: kernel-core-6.3.0-0.rc3.20230324git1e760fa3596e.33.fc39.x86_64                                                                                      
dkms: running auto installation service for kernel 6.3.0-0.rc3.20230324git1e760fa3596e.33.fc39.x86_64
Error! Could not locate dkms.conf file.
File: /var/lib/dkms/8192eu/5.11.2.1-4.git20230212.fc39/source/dkms.conf does not exist.
Done. 
dkms: running auto installation service for kernel 6.3.0-0.rc3.20230324git1e760fa3596e.33.fc39.x86_64
Error! Could not locate dkms.conf file.
File: /var/lib/dkms/8192eu/5.11.2.1-4.git20230212.fc39/source/dkms.conf does not exist.
Done. 
```

**Something is not right here**
```
$ dnf list --installed | grep dkms
dkms.noarch                                     3.0.10-3.fc38                               @rawhide                                              
dkms-rtl8192eu.noarch                           5.11.2.1-5.git20230312.fc39                 @copr:copr.fedorainfracloud.org:sunwire:dkms-rtl8192eu
```

Why is it not working?

onyeibo, 2023-03-28:
That worked!!

Wireless card now operational for Kernel 6.3.  Thank you!

closed 2023-03-28 by onyeibo