clnhub/rtl8192eu-linux #81: Can't enter monitor mode from Kali Linux

xn1llox, 2023-06-03:
it just keeps loading and gives me an error in the console

clnhub, 2023-06-08:
Did you enable monitor mode before building?

Rakhimiy, 2024-03-27:
me to i cant now enter to monitor mode

albaluz35, 2024-05-16:
In my case enabling monitor mode and doing packet injection WORKS! as follows:

sudo apt update
sudo apt upgrade
sudo apt-get install linux-headers-`uname -r` build-essential dkms
git clone https://github.com/clnhub/rtl8192eu-linux.git
cd rtl8192eu-linux
sed -i -E 's/CONFIG_WIFI_MONITOR\s*=\s*n/CONFIG_WIFI_MONITOR = y/' Makefile

sudo rmmod 8192eu rtl8xxxu r8188eu
echo "blacklist rtl8xxxu" >> ./blacklist-rtl8xxxu.conf
sudo mv ./blacklist-rtl8xxxu.conf /etc/modprobe.d/
./install_wifi.sh

-----------------------

**IF THERE ARE PROBLEMS WITH ./install_wifi.sh, RUN ./uninstall_wifi.sh then execute make commands manually:**
\# sudo make
\# sudo make install

-----------------

sudo modprobe 8192eu
sudo lsmod |grep 8192eu                                                     

!To enable Monitor mode and test packet injection:!
=================================================

WL=wlan0
sudo iwconfig
sudo ifconfig $WL down
sudo airmon-ng check kill
sudo iwconfig $WL mode monitor
sudo ifconfig $WL up
sudo iwconfig

sudo aireplay-ng --test $WL

closed 2025-01-02 by clnhub