clnhub/rtl8192eu-linux #82: Kernel 6.3.6 - Compiling error

raphaelquati, 2023-06-11:
After installing Fedora 38 (kernel 6.3.6-200.fc38.x86_64), I tried to compile (from master)

```
/home/rcouto/develop/rtl8192eu-linux/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_ch_switch_notify’:
/home/rcouto/develop/rtl8192eu-linux/os_dep/linux/ioctl_cfg80211.c:396:9: error: too few arguments to function ‘cfg80211_ch_switch_notify’
  396 |         cfg80211_ch_switch_notify(adapter->pnetdev, &chdef
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/rcouto/develop/rtl8192eu-linux/include/osdep_service_linux.h:93,
                 from /home/rcouto/develop/rtl8192eu-linux/include/osdep_service.h:47,
                 from /home/rcouto/develop/rtl8192eu-linux/include/drv_types.h:27,
                 from /home/rcouto/develop/rtl8192eu-linux/os_dep/linux/ioctl_cfg80211.c:17:
./include/net/cfg80211.h:8375:6: note: declared here
 8375 | void cfg80211_ch_switch_notify(struct net_device *dev,
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~
```

The kernel function was updated by https://github.com/torvalds/linux/commit/b345f0637c0042f9e6b78378a32256d90f485774

But I'm not a kernel driver expert, but I have made a small change to the code, to add the missing parameter. I have successfully tested with my TL-WN821N.

clnhub, 2023-06-12:
This was fixed on the (newer) default branch 5.11.2.1. Is it possible to use this branch?

raphaelquati, 2023-06-12:
It's working with 5.11.2.1 branch.