clnhub/rtl8192eu-linux #87: Symbol version dump "Module.symvers" is missing.

ffpp2003, 2023-08-05:
I'm trying to build this driver for a laptop with a WonderMedia WM8650 CPU (armel/armv5). I have modified the makefile to compile as a 32 bit arm, and I'm calling the make command this way:
`make ARCH=arm CROSS_COMPILE=arm-linux-gnueabi- KVER="5.15.64-g0ae752685214-dirty" KSRC="/home/fpp/Documentos/kernel-5.15/" -j6`
This are the errors I get:
```
WARNING: Symbol version dump "Module.symvers" is missing.
         Modules may not have dependencies or modversions.
         You may get many unresolved symbol warnings.
  MODPOST /home/fpp/Documentos/rtl8192eu-linux/Module.symvers
WARNING: modpost: "usb_register_driver" [/home/fpp/Documentos/rtl8192eu-linux/8192eu.ko] undefined!
WARNING: modpost: "__napi_schedule" [/home/fpp/Documentos/rtl8192eu-linux/8192eu.ko] undefined!
WARNING: modpost: "strsep" [/home/fpp/Documentos/rtl8192eu-linux/8192eu.ko] undefined!
WARNING: modpost: "strcasecmp" [/home/fpp/Documentos/rtl8192eu-linux/8192eu.ko] undefined!
WARNING: modpost: "eth_type_trans" [/home/fpp/Documentos/rtl8192eu-linux/8192eu.ko] undefined!
WARNING: modpost: "cfg80211_external_auth_request" [/home/fpp/Documentos/rtl8192eu-linux/8192eu.ko] undefined!
WARNING: modpost: "napi_gro_receive" [/home/fpp/Documentos/rtl8192eu-linux/8192eu.ko] undefined!
WARNING: modpost: "single_release" [/home/fpp/Documentos/rtl8192eu-linux/8192eu.ko] undefined!
WARNING: modpost: "__aeabi_uidiv" [/home/fpp/Documentos/rtl8192eu-linux/8192eu.ko] undefined!
WARNING: modpost: "__stack_chk_fail" [/home/fpp/Documentos/rtl8192eu-linux/8192eu.ko] undefined!
WARNING: modpost: suppressed 191 unresolved symbol warnings because there were too many)
make[3]: *** There are no rules to build the target 'scripts/module.lds', needed for '/home/fpp/Documentos/rtl8192eu-linux/8192eu.ko'.  Stop.
make[3]: *** Waiting for other tasks to finish....
  CC [M]  /home/fpp/Documentos/rtl8192eu-linux/8192eu.mod.o
make[2]: *** [scripts/Makefile.modpost:140: __modpost] Error 2
make[1]: *** [Makefile:1811: modules] Error 2
make[1]: exiting directory '/home/fpp/Documentos/kernel-5.15'
make: *** [Makefile:2459: modules] Error 2
```

clnhub, 2023-08-06:
The file should be part of the (ARM) distribution (kernel source). 

If it is not there, you may have to add it to your (cross compile) kernel source directory manually.

ffpp2003, 2023-08-06:
Thanks, I had forgotten to run `make modules` on the kernel source

closed 2023-08-06 by ffpp2003