clnhub/rtl8192eu-linux #88: ./install_wifi.sh error

MiChaelinzo, 2023-08-20:
I did `sudo apt install linux-headers-generic build-essential dkms git
` before installing too! Why it's using linux-headers-6.1.21-v8+ is there a version for 5.10.0-25-arm64 since that is the linux headers version for my generic build-essential?

```
rmmod: ERROR: Module 8192eu is not currently loaded
rmmod: ERROR: Module rtl8xxxu is not currently loaded
Error! The module rtl8192eu 1.0 is not currently installed.
This module is not currently ACTIVE for kernel 6.1.21-v8+ (aarch64).
Error! There is no instance of rtl8192eu 1.0
for kernel 6.1.21-v8+ (aarch64) located in the DKMS tree.

Creating symlink /var/lib/dkms/rtl8192eu/1.0/source ->
                 /usr/src/rtl8192eu-1.0

DKMS: add completed.
Error! Your kernel headers for kernel 6.1.21-v8+ cannot be found.
Please install the linux-headers-6.1.21-v8+ package,
or use the --kernelsourcedir option to tell DKMS where it's located
modprobe: FATAL: Module 8192eu not found in directory /lib/modules/6.1.21-v8+

```

davingthe, 2024-04-07:
i'm having a similar issue on eOS 7.1

davingthe, 2024-04-07:
I'm having a similar issue on eOS 7.1 (6.5.0-26-generic)

closed 2025-01-02 by clnhub