clnhub/rtl8192eu-linux #89: Dkms build install fail! I tried all fixes, but installation is not working!

MiChaelinzo, 2023-08-20:
I also tried `sudo apt-get git raspberrypi-kernel-headers build-essential dkms since I have Raspberry Pi 4 modebl B and 
...
CONFIG_PLATFORM_I386_PC = n
...
CONFIG_PLATFORM_ARM_RPI = y
````
DKMS make.log for rtl8192eu-1.0 for kernel 6.1.21-v8+ (aarch64)
Sun 20 Aug 2023 06:31:59 PM +03
make ARCH=arm CROSS_COMPILE= -C /lib/modules/6.1.21-v8+/build M=/var/lib/dkms/rtl8192eu/1.0/build  modules
make[1]: Entering directory '/usr/src/linux-headers-6.1.21-v8+'
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_security.o
gcc: error: unrecognized argument in option ‘-mabi=apcs-gnu’
gcc: note: valid arguments to ‘-mabi=’ are: ilp32 lp64
gcc: error: unrecognized argument in option ‘-mabi=apcs-gnu’
gcc: note: valid arguments to ‘-mabi=’ are: ilp32 lp64
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_debug.o
gcc: error: unrecognized argument in option ‘-mabi=apcs-gnu’
gcc: note: valid arguments to ‘-mabi=’ are: ilp32 lp64
gcc: error: unrecognized command-line option ‘-mapcs’
gcc: error: unrecognized command-line option ‘-mapcs’
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_io.o
gcc: error: unrecognized command-line option ‘-mapcs’
gcc: error: unrecognized argument in option ‘-mabi=apcs-gnu’
gcc: note: valid arguments to ‘-mabi=’ are: ilp32 lp64
gcc: error: unrecognized command-line option ‘-mno-sched-prolog’
gcc: error: unrecognized command-line option ‘-mapcs’
gcc: error: unrecognized command-line option ‘-mno-sched-prolog’
gcc: error: unrecognized command-line option ‘-msoft-float’
gcc: error: unrecognized command-line option ‘-mno-sched-prolog’
gcc: error: unrecognized command-line option ‘-msoft-float’
gcc: error: unrecognized command-line option ‘-msoft-float’
make[2]: *** [scripts/Makefile.build:250: /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_security.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [scripts/Makefile.build:250: /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_debug.o] Error 1
make[2]: *** [scripts/Makefile.build:250: /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_cmd.o] Error 1
gcc: error: unrecognized command-line option ‘-mno-sched-prolog’
gcc: error: unrecognized command-line option ‘-msoft-float’
make[2]: *** [scripts/Makefile.build:250: /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_io.o] Error 1
make[1]: *** [Makefile:2012: /var/lib/dkms/rtl8192eu/1.0/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.1.21-v8+'
make: *** [Makefile:2459: modules] Error 2

```

clnhub, 2023-08-20:
Please try `CONFIG_PLATFORM_ARM_AARCH64 = y`

MiChaelinzo, 2023-08-20:
Yeah I fixed it etc. but I cannot enable monitor mode with airmon-ng

MiChaelinzo, 2023-08-20:
It has this error 
```
PHY    Interface    Driver        Chipset

phy0    wlan0        brcmfmac    Broadcom 43430
phy1    wlan1        rtl8192eu    TP-Link TL-WN823N v2/v3 [Realtek RTL8192EU]
        (monitor mode enabled)

Error setting channel: command failed: Device or resource busy (-16)
Error -16 likely means your card was set back to station mode by something.
Removing non-monitor wlan1 interface...
        (monitor mode disabled) 
```

clnhub, 2025-01-02:
I am closing this.  This is not related to a driver problem, but more a question about running multiple chips and monitor mode. There are other (solved) issues about this.

closed 2025-01-02 by clnhub