clnhub/rtl8192eu-linux #92: Failing for Kernel-6.7.0-0.rc3.* kernels on Fedora

onyeibo, 2023-11-30:
See dnf output during update:

```
Running scriptlet: dkms-3.0.12-1.fc40.noarch                                                                                               6/6 
Created symlink /etc/systemd/system/multi-user.target.wants/dkms.service → /usr/lib/systemd/system/dkms.service.

  Running scriptlet: kernel-modules-core-6.7.0-0.rc3.20231128gitdf60cee26a2e.29.fc40.x86_64                                                  6/6 
  Running scriptlet: kernel-core-6.7.0-0.rc3.20231128gitdf60cee26a2e.29.fc40.x86_64                                                          6/6 
dkms: running auto installation service for kernel 6.7.0-0.rc3.20231128gitdf60cee26a2e.29.fc40.x86_64Sign command: /lib/modules/6.7.0-0.rc3.20231128gitdf60cee26a2e.29.fc40.x86_64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
'make' -j8 all KVER=6.7.0-0.rc3.20231128gitdf60cee26a2e.29.fc40.x86_64........(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.7.0-0.rc3.20231128gitdf60cee26a2e.29.fc40.x86_64 (x86_64)
Consult /var/lib/dkms/rtl8192eu/1.0/build/make.log for more information.
dkms autoinstall on 6.7.0-0.rc3.20231128gitdf60cee26a2e.29.fc40.x86_64/x86_64 failed for rtl8192eu(10)
Error! One or more modules failed to install during autoinstall.
Refer to previous errors for more information.
 Failed. 
dkms: running auto installation service for kernel 6.7.0-0.rc3.20231128gitdf60cee26a2e.29.fc40.x86_64Sign command: /lib/modules/6.7.0-0.rc3.20231128gitdf60cee26a2e.29.fc40.x86_64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
'make' -j8 all KVER=6.7.0-0.rc3.20231128gitdf60cee26a2e.29.fc40.x86_64........(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.7.0-0.rc3.20231128gitdf60cee26a2e.29.fc40.x86_64 (x86_64)
Consult /var/lib/dkms/rtl8192eu/1.0/build/make.log for more information.
dkms autoinstall on 6.7.0-0.rc3.20231128gitdf60cee26a2e.29.fc40.x86_64/x86_64 failed for rtl8192eu(10)
Error! One or more modules failed to install during autoinstall.
Refer to previous errors for more information.
 Failed. 

  Running scriptlet: dkms-3.0.12-1.fc40.noarch   
```

Module is working on same machine while running kernel 6.6.0-61.fc40

sunwire, 2023-11-30:
Do you use the latest code from the git repo? [This](https://github.com/clnhub/rtl8192eu-linux/commit/9b190cf9797261172fb36cdbf9a7ae6ff0ec3c06) commit adds support for the kernel 6.7.x

onyeibo, 2023-12-03:
> Do you use the latest code from the git repo? [This](https://github.com/clnhub/rtl8192eu-linux/commit/9b190cf9797261172fb36cdbf9a7ae6ff0ec3c06) commit adds support for the kernel 6.7.x

I am just realizing that the copr repo providing this driver is no longer on the machine.  Please can you share the URL to the repo again?

onyeibo, 2023-12-04:
> > lease can you share the URL to the repo again?
> 
> https://copr.fedorainfracloud.org/coprs/sunwire/dkms-rtl8192eu/

I installed dkms-rtl8192eu from this repo.
```
# dnf list --installed | grep dkms
dkms.noarch                                    3.0.12-1.fc40                               @rawhide                                              
dkms-rtl8192eu.noarch                          5.11.2.1-5.git20231130.fc40                 @copr:copr.fedorainfracloud.org:sunwire:dkms-rtl8192eu
```
The wireless driver still did not compile.  Here is the DNF output:

```
Running scriptlet: kernel-core-6.7.0-0.rc3.20231201git994d5c58e50e.32.fc40.x86_64                                                          6/6
dkms: running auto installation service for kernel 6.7.0-0.rc3.20231201git994d5c58e50e.32.fc40.x86_64Sign command: /lib/modules/6.7.0-0.rc3.20231201git994d5c58e50e.32.fc40.x86_64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
'make' -j8 KVER=6.7.0-0.rc3.20231201git994d5c58e50e.32.fc40.x86_64............
Signing module /var/lib/dkms/8192eu/5.11.2.1-5.git20231130.fc40/build/8192eu.ko
Cleaning build area...

8192eu.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.7.0-0.rc3.20231201git994d5c58e50e.32.fc40.x86_64/extra/
depmod...
Warning: The unit file, source configuration file or drop-ins of systemd-modules-load.service changed on disk. Run 'systemctl daemon-reload' to reload units.
Sign command: /lib/modules/6.7.0-0.rc3.20231201git994d5c58e50e.32.fc40.x86_64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
'make' -j8 all KVER=6.7.0-0.rc3.20231201git994d5c58e50e.32.fc40.x86_64.......(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.7.0-0.rc3.20231201git994d5c58e50e.32.fc40.x86_64 (x86_64)
Consult /var/lib/dkms/rtl8192eu/1.0/build/make.log for more information.
dkms autoinstall on 6.7.0-0.rc3.20231201git994d5c58e50e.32.fc40.x86_64/x86_64 succeeded for 8192eu
dkms autoinstall on 6.7.0-0.rc3.20231201git994d5c58e50e.32.fc40.x86_64/x86_64 failed for rtl8192eu(10)
Error! One or more modules failed to install during autoinstall.
Refer to previous errors for more information.
 Failed.
dkms: running auto installation service for kernel 6.7.0-0.rc3.20231201git994d5c58e50e.32.fc40.x86_64Sign command: /lib/modules/6.7.0-0.rc3.20231201git994d5c58e50e.32.fc40.x86_64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
'make' -j8 all KVER=6.7.0-0.rc3.20231201git994d5c58e50e.32.fc40.x86_64.......(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.7.0-0.rc3.20231201git994d5c58e50e.32.fc40.x86_64 (x86_64)
Consult /var/lib/dkms/rtl8192eu/1.0/build/make.log for more information.
dkms autoinstall on 6.7.0-0.rc3.20231201git994d5c58e50e.32.fc40.x86_64/x86_64 succeeded for 8192eu
dkms autoinstall on 6.7.0-0.rc3.20231201git994d5c58e50e.32.fc40.x86_64/x86_64 failed for rtl8192eu(10)
Error! One or more modules failed to install during autoinstall.
Refer to previous errors for more information.
 Failed.

  Running scriptlet: dkms-rtl8192eu-5.11.2.1-5.git20231130.fc40.noarch
```

sunwire, 2023-12-04:
> The wireless driver still did not compile. Here is the DNF output:

I think something is wrong with your OS

> Consult /var/lib/dkms/rtl8192eu/1.0/build/make.log for more information.

For me it looks like you didn't remove an old code because version should be 5.11.2.1-5.git20231130.fc40 not 1.0
and now you have two versions the old one 1.0 (probably from git) and the new one 5.11.2.1-5.git20231130.fc40 from COPR

```
[root@fedora ~]# rpm -q dkms dkms-rtl8192eu 
dkms-3.0.12-1.fc40.noarch
dkms-rtl8192eu-5.11.2.1-5.git20231130.fc40.noarch

[root@fedora ~]# dkms status
8192eu/5.11.2.1-5.git20231130.fc40, 6.7.0-0.rc3.20231201git994d5c58e50e.32.fc40.x86_64, x86_64: installed

[root@fedora ~]# modprobe 8192eu 

[root@fedora ~]# lsmod|grep  8192eu 
8192eu               2236416  0
cfg80211             1335296  1 8192eu

[root@fedora ~]# modinfo 8192eu 
filename:       /lib/modules/6.7.0-0.rc3.20231201git994d5c58e50e.32.fc40.x86_64/extra/8192eu.ko.xz
version:        v5.11.2.1-18-g8e7df912b.20210527_COEX20171113-0047
name:           8192eu
vermagic:       6.7.0-0.rc3.20231201git994d5c58e50e.32.fc40.x86_64 SMP preempt mod_unload 
sig_id:         PKCS#7
signer:         DKMS module signing key
```

sunwire, 2023-12-04:
> What do you propose I do?

Choose one, COPR or git (but please remember to pull changes from git so that you always use the latest code)

closed 2023-12-07 by onyeibo