clnhub/rtl8192eu-linux #100: dkms autoinstall fails for 6.9.0-0.* kernels on Fedora

onyeibo, 2024-03-29:
dkms: running auto installation service for kernel 6.9.0-0.rc1.20240326git928a87efa423.17.fc41.x86_64
Error! rtl8192eu/1.0 is broken! Missing the source directory or the symbolic link pointing to it.
Manual intervention is required!
Sign command: /lib/modules/6.9.0-0.rc1.20240326git928a87efa423.17.fc41.x86_64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
Building module(s)........(bad exit status: 2)
Failed command:
'make' -j8 KVER=6.9.0-0.rc1.20240326git928a87efa423.17.fc41.x86_64
Error! Bad return status for module build on kernel: 6.9.0-0.rc1.20240326git928a87efa423.17.fc41.x86_64 (x86_64)
Consult /var/lib/dkms/8192eu/5.11.2.1-5.git20240210.fc40/build/make.log for more information.
dkms autoinstall on 6.9.0-0.rc1.20240326git928a87efa423.17.fc41.x86_64/x86_64 failed for 8192eu(10)
Error! One or more modules failed to install during autoinstall.
Refer to previous errors for more information.
dkms: autoinstall for kernel 6.9.0-0.rc1.20240326git928a87efa423.17.fc41.x86_64 Failed. 
dkms: running auto installation service for kernel 6.9.0-0.rc1.20240326git928a87efa423.17.fc41.x86_64
Error! rtl8192eu/1.0 is broken! Missing the source directory or the symbolic link pointing to it.
Manual intervention is required!
Sign command: /lib/modules/6.9.0-0.rc1.20240326git928a87efa423.17.fc41.x86_64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

sunwire, 2024-03-29:
You are using the COPR version so try to update to the latest version and try again.

onyeibo, 2024-03-30:
> You are using the COPR version so try to update to the latest version and try again.

Ah ... that worked.
dnf didn't pull that recent driver initially.  Thanks

closed 2024-03-30 by onyeibo

sunwire, 2024-03-30:
> dnf didn't pull that recent driver initially.

No, I only updated the package after your report. For some reason I didn't receive a notification about a new commit from clnhub github.

BTW. https://fedoramagazine.org/cve-2024-3094-security-alert-f40-rawhide/

onyeibo, 2024-03-31:
> > dnf didn't pull that recent driver initially.
> 
> No, I only updated the package after your report. For some reason I didn't receive a notification about a new commit from clnhub github.
> 
> BTW. https://fedoramagazine.org/cve-2024-3094-security-alert-f40-rawhide/

Glad the report was useful.  I saw the CVE alert on LinkedIn yesterday.  Rawhide has gone back to xz-5.4.x already.  I checked my instance.  I should probably run another update, just in case