clnhub/rtl8192eu-linux #102: Monitor mode not supported

hossamfrhat, 2024-05-10:
it gave me an error when tried to enable monitor mode in parrot OS 
[image]

hossamfrhat, 2024-05-10:
i tried the driver on kali linux (2024.1) and it gave the same error

clnhub, 2024-05-11:
Did you enable it before building?

albaluz35, 2024-05-16:
In my case enabling monitor mode and doing packet injection WORKS! as follows:

sudo apt update
sudo apt upgrade
sudo apt-get install linux-headers-`uname -r` build-essential dkms
git clone https://github.com/clnhub/rtl8192eu-linux.git
cd rtl8192eu-linux
sed -i -E 's/CONFIG_WIFI_MONITOR\s*=\s*n/CONFIG_WIFI_MONITOR = y/' Makefile

sudo rmmod 8192eu rtl8xxxu r8188eu
echo "blacklist rtl8xxxu" >> ./blacklist-rtl8xxxu.conf
sudo mv ./blacklist-rtl8xxxu.conf /etc/modprobe.d/
./install_wifi.sh
################################################# 
# IF THERE ARE PROBLEMS WITH ./install_wifi.sh, RUN ./uninstall_wifi.sh
# then execute make commands manually:
# sudo make
# sudo make install
#################################################
sudo modprobe 8192eu
sudo lsmod |grep 8192eu                                                     

!To enable Monitor mode and test packet injection:!
=================================================
WL=wlan0
sudo iwconfig
sudo ifconfig $WL down
sudo airmon-ng check kill
sudo iwconfig $WL mode monitor
sudo ifconfig $WL up
sudo iwconfig

sudo aireplay-ng --test $WL

clnhub, 2025-01-02:
This seems to be more related to how to use monitor mode than to an issue with the code.

If the issue is driver related after all, please re-open.

Issue  [#112](https://github.com/clnhub/rtl8192eu-linux/issues/112) has a good manual on how to use monitor mode.

closed 2025-01-02 by clnhub