clnhub/rtl8192eu-linux #103: Unable to install driver on Jetson TK1

stm32h757, 2024-08-03:
Hello. I followed your official guide and result was this :

ubuntu@tegra-ubuntu:~/Desktop/WirelessDriver/rtl8192eu-linux-driver$ make
make ARCH=arm CROSS_COMPILE= -C /lib/modules/3.10.40-ga7da876/build M=/home/ubuntu/Desktop/WirelessDriver/rtl8192eu-linux-driver  modules
make[1]: Entering directory `/usr/src/linux-headers-3.10.40-ga7da876'
  CC [M]  /home/ubuntu/Desktop/WirelessDriver/rtl8192eu-linux-driver/core/rtw_cmd.o
  CC [M]  /home/ubuntu/Desktop/WirelessDriver/rtl8192eu-linux-driver/core/rtw_security.o
  CC [M]  /home/ubuntu/Desktop/WirelessDriver/rtl8192eu-linux-driver/core/rtw_debug.o
  CC [M]  /home/ubuntu/Desktop/WirelessDriver/rtl8192eu-linux-driver/core/rtw_io.o
  CC [M]  /home/ubuntu/Desktop/WirelessDriver/rtl8192eu-linux-driver/core/rtw_ioctl_set.o
  CC [M]  /home/ubuntu/Desktop/WirelessDriver/rtl8192eu-linux-driver/core/rtw_ieee80211.o
/home/ubuntu/Desktop/WirelessDriver/rtl8192eu-linux-driver/core/rtw_ieee80211.c: In function ‘dump_ht_cap_ie’:
/home/ubuntu/Desktop/WirelessDriver/rtl8192eu-linux-driver/core/rtw_ieee80211.c:1652:24: warning: ‘ht_cap_ielen’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  dump_ht_cap_ie_content(sel, ht_cap_ie + 2, ht_cap_ielen);
                        ^
/home/ubuntu/Desktop/WirelessDriver/rtl8192eu-linux-driver/core/rtw_ieee80211.c: In function ‘dump_ht_op_ie’:
/home/ubuntu/Desktop/WirelessDriver/rtl8192eu-linux-driver/core/rtw_ieee80211.c:1685:23: warning: ‘ht_op_ielen’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  dump_ht_op_ie_content(sel, ht_op_ie + 2, ht_op_ielen);
                       ^
  CC [M]  /home/ubuntu/Desktop/WirelessDriver/rtl8192eu-linux-driver/core/rtw_mlme.o
  CC [M]  /home/ubuntu/Desktop/WirelessDriver/rtl8192eu-linux-driver/core/rtw_mlme_ext.o
/home/ubuntu/Desktop/WirelessDriver/rtl8192eu-linux-driver/core/rtw_mlme_ext.c:76:3: error: ‘WLAN_CATEGORY_RADIO_MEASUREMENT’ undeclared here (not in a function)
  {WLAN_CATEGORY_RADIO_MEASUREMENT, "ACTION_RADIO_MEAS", &on_action_rm},
   ^
/home/ubuntu/Desktop/WirelessDriver/rtl8192eu-linux-driver/core/rtw_mlme_ext.c:87:3: error: ‘WLAN_CATEGORY_WNM_UNPROTECTED’ undeclared here (not in a function)
  {WLAN_CATEGORY_WNM_UNPROTECTED, "ACTION_UNPROTECTED_WNM", &DoReserved},
   ^
/home/ubuntu/Desktop/WirelessDriver/rtl8192eu-linux-driver/core/rtw_mlme_ext.c: In function ‘rtw_txpwr_init_regd’:
/home/ubuntu/Desktop/WirelessDriver/rtl8192eu-linux-driver/core/rtw_mlme_ext.c:261:3: warning: empty declaration [enabled by default]
   __attribute__((__fallthrough__));
   ^
/home/ubuntu/Desktop/WirelessDriver/rtl8192eu-linux-driver/core/rtw_mlme_ext.c:261:3: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
/home/ubuntu/Desktop/WirelessDriver/rtl8192eu-linux-driver/core/rtw_mlme_ext.c: In function ‘issue_action_BSSCoexistPacket’:
/home/ubuntu/Desktop/WirelessDriver/rtl8192eu-linux-driver/core/rtw_mlme_ext.c:10499:33: error: ‘WLAN_EID_BSS_INTOLERANT_CHL_REPORT’ undeclared (first use in this function)
     pframe = rtw_set_ie(pframe, WLAN_EID_BSS_INTOLERANT_CHL_REPORT, k, InfoContent, &(pattrib->pktlen));
                                 ^
/home/ubuntu/Desktop/WirelessDriver/rtl8192eu-linux-driver/core/rtw_mlme_ext.c:10499:33: note: each undeclared identifier is reported only once for each function it appears in
/home/ubuntu/Desktop/WirelessDriver/rtl8192eu-linux-driver/core/rtw_mlme_ext.c:10499:5: warning: passing argument 2 of ‘rtw_set_ie’ makes integer from pointer without a cast [enabled by default]
     pframe = rtw_set_ie(pframe, WLAN_EID_BSS_INTOLERANT_CHL_REPORT, k, InfoContent, &(pattrib->pktlen));
     ^
In file included from /home/ubuntu/Desktop/WirelessDriver/rtl8192eu-linux-driver/include/drv_types.h:31:0,
                 from /home/ubuntu/Desktop/WirelessDriver/rtl8192eu-linux-driver/core/rtw_mlme_ext.c:17:
/home/ubuntu/Desktop/WirelessDriver/rtl8192eu-linux-driver/include/ieee80211.h:1030:5: note: expected ‘sint’ but argument is of type ‘struct action_handler *’
 u8 *rtw_set_ie(u8 *pbuf, sint index, uint len, const u8 *source, uint *frlen);
     ^
make[2]: *** [/home/ubuntu/Desktop/WirelessDriver/rtl8192eu-linux-driver/core/rtw_mlme_ext.o] Error 1
make[1]: *** [_module_/home/ubuntu/Desktop/WirelessDriver/rtl8192eu-linux-driver] Error 2
make[1]: Leaving directory `/usr/src/linux-headers-3.10.40-ga7da876'
make: *** [modules] Error 2
ubuntu@tegra-ubuntu:~/Desktop/WirelessDriver/rtl8192eu-linux-driver$ sudo make
make ARCH=arm CROSS_COMPILE= -C /lib/modules/3.10.40-ga7da876/build M=/home/ubuntu/Desktop/WirelessDriver/rtl8192eu-linux-driver  modules
make[1]: Entering directory `/usr/src/linux-headers-3.10.40-ga7da876'
  CC [M]  /home/ubuntu/Desktop/WirelessDriver/rtl8192eu-linux-driver/core/rtw_mlme_ext.o
/home/ubuntu/Desktop/WirelessDriver/rtl8192eu-linux-driver/core/rtw_mlme_ext.c:76:3: error: ‘WLAN_CATEGORY_RADIO_MEASUREMENT’ undeclared here (not in a function)
  {WLAN_CATEGORY_RADIO_MEASUREMENT, "ACTION_RADIO_MEAS", &on_action_rm},
   ^
/home/ubuntu/Desktop/WirelessDriver/rtl8192eu-linux-driver/core/rtw_mlme_ext.c:87:3: error: ‘WLAN_CATEGORY_WNM_UNPROTECTED’ undeclared here (not in a function)
  {WLAN_CATEGORY_WNM_UNPROTECTED, "ACTION_UNPROTECTED_WNM", &DoReserved},
   ^
/home/ubuntu/Desktop/WirelessDriver/rtl8192eu-linux-driver/core/rtw_mlme_ext.c: In function ‘rtw_txpwr_init_regd’:
/home/ubuntu/Desktop/WirelessDriver/rtl8192eu-linux-driver/core/rtw_mlme_ext.c:261:3: warning: empty declaration [enabled by default]
   __attribute__((__fallthrough__));
   ^
/home/ubuntu/Desktop/WirelessDriver/rtl8192eu-linux-driver/core/rtw_mlme_ext.c:261:3: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
/home/ubuntu/Desktop/WirelessDriver/rtl8192eu-linux-driver/core/rtw_mlme_ext.c: In function ‘issue_action_BSSCoexistPacket’:
/home/ubuntu/Desktop/WirelessDriver/rtl8192eu-linux-driver/core/rtw_mlme_ext.c:10499:33: error: ‘WLAN_EID_BSS_INTOLERANT_CHL_REPORT’ undeclared (first use in this function)
     pframe = rtw_set_ie(pframe, WLAN_EID_BSS_INTOLERANT_CHL_REPORT, k, InfoContent, &(pattrib->pktlen));
                                 ^
/home/ubuntu/Desktop/WirelessDriver/rtl8192eu-linux-driver/core/rtw_mlme_ext.c:10499:33: note: each undeclared identifier is reported only once for each function it appears in
/home/ubuntu/Desktop/WirelessDriver/rtl8192eu-linux-driver/core/rtw_mlme_ext.c:10499:5: warning: passing argument 2 of ‘rtw_set_ie’ makes integer from pointer without a cast [enabled by default]
     pframe = rtw_set_ie(pframe, WLAN_EID_BSS_INTOLERANT_CHL_REPORT, k, InfoContent, &(pattrib->pktlen));
     ^
In file included from /home/ubuntu/Desktop/WirelessDriver/rtl8192eu-linux-driver/include/drv_types.h:31:0,
                 from /home/ubuntu/Desktop/WirelessDriver/rtl8192eu-linux-driver/core/rtw_mlme_ext.c:17:
/home/ubuntu/Desktop/WirelessDriver/rtl8192eu-linux-driver/include/ieee80211.h:1030:5: note: expected ‘sint’ but argument is of type ‘struct action_handler *’
 u8 *rtw_set_ie(u8 *pbuf, sint index, uint len, const u8 *source, uint *frlen);
     ^
make[2]: *** [/home/ubuntu/Desktop/WirelessDriver/rtl8192eu-linux-driver/core/rtw_mlme_ext.o] Error 1
make[1]: *** [_module_/home/ubuntu/Desktop/WirelessDriver/rtl8192eu-linux-driver] Error 2
make[1]: Leaving directory `/usr/src/linux-headers-3.10.40-ga7da876'
make: *** [modules] Error 2
ubuntu@tegra-ubuntu:~/Desktop/WirelessDriver/rtl8192eu-linux-driver$ 

I'm using Jetson TK1,
so linux version is Linux_for_Tegra 21.8.0(actually it is ubuntu 14.04)
and kernel version is 3.10.40-ga7da876

stm32h757, 2024-08-03:
ubuntu@tegra-ubuntu:~/Desktop/WirelessDriver/linux_v4.3.1.1_11320$ sudo sh ./install.sh
##################################################
Realtek Wi-Fi driver Auto installation script
Novembor, 21 2011 v1.1.0
##################################################
Decompress the driver source tar ball:
	rtl8192EU_linux_v4.3.1.1_11320.20140505.tar.gz
rtl8192EU_linux_v4.3.1.1_11320.20140505/
rtl8192EU_linux_v4.3.1.1_11320.20140505/clean
rtl8192EU_linux_v4.3.1.1_11320.20140505/core/
rtl8192EU_linux_v4.3.1.1_11320.20140505/core/efuse/
rtl8192EU_linux_v4.3.1.1_11320.20140505/core/efuse/rtw_efuse.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/core/rtw_ap.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/core/rtw_beamforming.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/core/rtw_br_ext.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/core/rtw_btcoex.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/core/rtw_bt_mp.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/core/rtw_cmd.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/core/rtw_debug.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/core/rtw_eeprom.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/core/rtw_ieee80211.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/core/rtw_io.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/core/rtw_ioctl_query.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/core/rtw_ioctl_rtl.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/core/rtw_ioctl_set.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/core/rtw_iol.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/core/rtw_mlme.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/core/rtw_mlme_ext.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/core/rtw_mp.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/core/rtw_mp_ioctl.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/core/rtw_odm.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/core/rtw_p2p.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/core/rtw_pwrctrl.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/core/rtw_recv.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/core/rtw_rf.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/core/rtw_security.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/core/rtw_sreset.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/core/rtw_sta_mgt.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/core/rtw_tdls.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/core/rtw_vht.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/core/rtw_wapi.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/core/rtw_wapi_sms4.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/core/rtw_wlan_util.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/core/rtw_xmit.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/hal/
rtl8192EU_linux_v4.3.1.1_11320.20140505/hal/HalPwrSeqCmd.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/hal/hal_btcoex.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/hal/hal_com.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/hal/hal_com_phycfg.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/hal/hal_hci/
rtl8192EU_linux_v4.3.1.1_11320.20140505/hal/hal_hci/hal_usb.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/hal/hal_intf.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/hal/hal_phy.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/hal/led/
rtl8192EU_linux_v4.3.1.1_11320.20140505/hal/led/hal_usb_led.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/hal/OUTSRC/
rtl8192EU_linux_v4.3.1.1_11320.20140505/hal/OUTSRC/HalPhyRf.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/hal/OUTSRC/HalPhyRf.h
rtl8192EU_linux_v4.3.1.1_11320.20140505/hal/OUTSRC/odm.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/hal/OUTSRC/odm.h
rtl8192EU_linux_v4.3.1.1_11320.20140505/hal/OUTSRC/odm_AntDiv.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/hal/OUTSRC/odm_AntDiv.h
rtl8192EU_linux_v4.3.1.1_11320.20140505/hal/OUTSRC/odm_debug.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/hal/OUTSRC/odm_debug.h
rtl8192EU_linux_v4.3.1.1_11320.20140505/hal/OUTSRC/odm_HWConfig.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/hal/OUTSRC/odm_HWConfig.h
rtl8192EU_linux_v4.3.1.1_11320.20140505/hal/OUTSRC/odm_interface.c
rtl8192EU_linux_v4.3.1.1_11320.20140505/hal/OUTSRC/odm_interface.h
rtl8192EU_linux_v4.3.1.1_11320.20140505/hal/OUTSRC/odm_precomp.h
rtl8192EU_linux_v4.3.1.1_11320.20140505/hal/OUTSRC/odm_reg.h
[... 297 lines cut ...]
cd hal/OUTSRC/ ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko 
cd hal/led ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd hal ; rm -fr */*/*.mod.c */*/*.mod */*/*.o */*/.*.cmd */*/*.ko
cd hal ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko
cd hal ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd core/efuse ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd core ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd os_dep/linux ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd os_dep ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd platform ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
rm -fr Module.symvers ; rm -fr Module.markers ; rm -fr modules.order
rm -fr *.mod.c *.mod *.o .*.cmd *.ko *~
rm -fr .tmp_versions
Authentication requested [root] for make driver:
./install.sh: 48: [: unexpected operator
make ARCH=armv7l CROSS_COMPILE= -C /lib/modules/3.10.40-ga7da876/build M=/home/ubuntu/Desktop/WirelessDriver/linux_v4.3.1.1_11320/driver/rtl8192EU_linux_v4.3.1.1_11320.20140505  modules
make[1]: Entering directory `/usr/src/linux-headers-3.10.40-ga7da876'
Makefile:587: /usr/src/linux-headers-3.10.40-ga7da876/arch/armv7l/Makefile: No such file or directory
make[1]: *** No rule to make target `/usr/src/linux-headers-3.10.40-ga7da876/arch/armv7l/Makefile'.  Stop.
make[1]: Leaving directory `/usr/src/linux-headers-3.10.40-ga7da876'
make: *** [modules] Error 2
##################################################
Compile make driver error: 2
Please check error Mesg
##################################################

clnhub, 2024-08-08:
The driver supports kernel 5 or higher.

closed 2025-01-02 by clnhub