clnhub/rtl8192eu-linux #111: Missing /proc/net/wireless while using rtl8192eu on Fedora

onyeibo, 2024-12-25:
I am running a minimal version of Fedora (rawhide) with the `rtl8192eu` driver from `copr:copr.fedorainfracloud.org:sunwire:dkms-rtl8192eu.`  I use `Bumblebee--status` for status bar while using `Sway` Windows Manager.  `Bumblebee-status` accesses `/proc/net/wireless` to display the state of wireless connection on the status bar.  Everything worked fine until recently.  Things changes after some updates weeks ago.

Bumblebee-status now crashes with the following message:
`"[Errno 2] No such file or directory: "proc/net/wireless`"  

I filed a bug report against Bumblebee-status here:
https://github.com/tobi-wan-kenobi/bumblebee-status/issues/1029

The developers believe this is a driver issue.  Note that the wireless device works -- I am able to connect via` nmcli` and see info via` ip addr` but `/proc/net/wireless` is truly absent.  Bumblebee-status resumes when I disable the nic module (responsible for getting network states)

clnhub, 2024-12-25:
This seems to be related to how the kernel was compiled. Did the failure start with an updated kernel?

Could you post `iwconfig --version` output?

And could you try with the latest available kernel.

sunwire, 2024-12-25:
This seems to be a problem with the latest kernel in rawhide 6.13 and not with the bumblebee-status or the driver itself.
I checked and the file /proc/net/wireless is indeed missing.

onyeibo, 2024-12-26:
Filed a report against the Kernel at:
https://bugzilla.redhat.com/show_bug.cgi?id=2334171

clnhub, 2024-12-26:
What I can find is that wireless extensions support (including /proc/net/wireless) has been removed from Fedora.

The build log (Retire wireless Extensions) is here: [https://koji.fedoraproject.org/koji/buildinfo?buildID=2586129](https://koji.fedoraproject.org/koji/buildinfo?buildID=2586129)

I cannot find any issues with the driver code related to this issue.

clnhub, 2025-01-02:
I am closing this, because removing a deprecated feature/API from a distribution is not related to the kernel driver. The driver uses the more modern cfg80211 kernel API.

The wiki is here: [https://fedoraproject.org/wiki/Changes/RemoveWirelessExtensions](https://fedoraproject.org/wiki/Changes/RemoveWirelessExtensions)
The discussion can be followed here : [https://discussion.fedoraproject.org/t/kernel-6-12-which-is-now-in-testing-is-missing-config-cfg80211-wext/138947](https://discussion.fedoraproject.org/t/kernel-6-12-which-is-now-in-testing-is-missing-config-cfg80211-wext/138947)

Possible solutions are:

1. Try a different distribution with Wireless Extensions enabled
2. Compile a Fedora kernel with `CONFIG_CFG80211_WEXT` enabled
3. Update software that still uses Wireless Extensions

closed 2025-01-02 by clnhub

nullr0ute, 2025-02-23:
> Possible solutions are:

4: Fix Bumblebee--status  to use the modern (but over 20 years old) libnl interface to get wireless status