clnhub/rtl8192eu-linux #112: Not working on Linux Mint 22 kernel version 6.8.0-51-generic

enelson1001, 2024-12-29:
**System Information**
```
System:
  Kernel: 6.8.0-51-generic arch: x86_64 bits: 64 compiler: gcc v: 13.3.0 clocksource: tsc
  Desktop: Cinnamon v: 6.2.9 tk: GTK v: 3.24.41 wm: Muffin v: 6.2.0 vt: 7 dm: LightDM v: 1.30.0
    Distro: Linux Mint 22 Wilma base: Ubuntu 24.04 noble
Machine:
  Type: Laptop System: Acer product: AO756 v: V1.02 serial: <superuser required> Chassis: type: 10
    v: 1.02 serial: <superuser required>
  Mobo: Acer model: Mimic v: Type2 - Board Version serial: <superuser required>
    part-nu: Type1Sku0 uuid: <superuser required> UEFI-[Legacy]: Acer v: 1.02 date: 04/26/2012
Battery:
  ID-1: BAT1 charge: 21.1 Wh (100.0%) condition: 21.1/37.0 Wh (57.2%) volts: 17.0 min: 14.8
    model: SANYO Li_Ion_4000mA type: Li-ion serial: <filter> status: full
CPU:
  Info: dual core model: Intel Celeron 877 bits: 64 type: MCP smt: <unsupported> arch: Sandy Bridge
    rev: 7 cache: L1: 128 KiB L2: 512 KiB L3: 2 MiB
  Speed (MHz): avg: 798 min/max: 800/1400 cores: 1: 798 2: 798 bogomips: 5587
  Flags: ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx
Graphics:
  Device-1: Intel 2nd Generation Core Processor Family Integrated Graphics
    vendor: Acer Incorporated ALI driver: i915 v: kernel arch: Gen-6 ports: active: LVDS-1
    empty: DP-1,HDMI-A-1,VGA-1 bus-ID: 00:02.0 chip-ID: 8086:0106 class-ID: 0300
  Device-2: Suyin HD WebCam driver: uvcvideo type: USB rev: 2.0 speed: 480 Mb/s lanes: 1
    bus-ID: 1-1.3:3 chip-ID: 064e:d251 class-ID: 0e02 serial: <filter>
  Display: x11 server: X.Org v: 21.1.11 with: Xwayland v: 23.2.6 driver: X: loaded: modesetting
    unloaded: fbdev,vesa dri: crocus gpu: i915 display-ID: :0 screens: 1
  Screen-1: 0 s-res: 1366x768 s-dpi: 96 s-size: 361x203mm (14.21x7.99") s-diag: 414mm (16.31")
  Monitor-1: LVDS-1 model: ChiMei InnoLux 0x1119 res: 1366x768 hz: 60 dpi: 136
    size: 256x144mm (10.08x5.67") diag: 294mm (11.6") modes: 1366x768
  API: EGL v: 1.5 hw: drv: intel crocus platforms: device: 0 drv: crocus device: 1 drv: swrast
    surfaceless: drv: crocus x11: drv: crocus inactive: gbm,wayland
  API: OpenGL v: 4.5 compat-v: 3.3 vendor: intel mesa v: 24.0.9-0ubuntu0.3 glx-v: 1.4
    direct-render: yes renderer: Mesa Intel HD Graphics 2000 (SNB GT1) device-ID: 8086:0106
Audio:
  Device-1: Intel 7 Series/C216 Family High Definition Audio vendor: Acer Incorporated ALI 7
    driver: snd_hda_intel v: kernel bus-ID: 00:1b.0 chip-ID: 8086:1e20 class-ID: 0403
  API: ALSA v: k6.8.0-51-generic status: kernel-api
  Server-1: PipeWire v: 1.0.5 status: active with: 1: pipewire-pulse status: active
    2: wireplumber status: active 3: pipewire-alsa type: plugin
Network:
  Device-1: Broadcom BCM4313 802.11bgn Wireless Network Adapter vendor: Foxconn driver: wl
    v: kernel pcie: speed: 2.5 GT/s lanes: 1 bus-ID: 03:00.0 chip-ID: 14e4:4727 class-ID: 0280
  IF: wlp3s0 state: up mac: <filter>
  Device-2: Broadcom NetLink BCM57785 Gigabit Ethernet PCIe vendor: Acer Incorporated ALI
    driver: tg3 v: kernel pcie: speed: 2.5 GT/s lanes: 1 port: N/A bus-ID: 04:00.0 chip-ID: 14e4:16b5
    class-ID: 0200
  IF: enp4s0f0 state: down mac: <filter>
  Device-3: TP-Link TL-WN823N v2/v3 [Realtek RTL8192EU] driver: rtl8192eu type: USB rev: 2.1
    speed: 480 Mb/s lanes: 1 bus-ID: 2-1.2:3 chip-ID: 2357:0109 class-ID: 0000 serial: <filter>
  IF: wlx98482768db02 state: down mac: <filter>
Drives:
  Local Storage: total: 465.76 GiB used: 13.36 GiB (2.9%)
  ID-1: /dev/sda vendor: Hitachi model: HTS545050A7E380 size: 465.76 GiB speed: 3.0 Gb/s
    tech: HDD rpm: 5400 serial: <filter> fw-rev: A6C0 scheme: GPT
Partition:
  ID-1: / size: 456.88 GiB used: 13.35 GiB (2.9%) fs: ext4 dev: /dev/sda3
  ID-2: /boot/efi size: 512 MiB used: 6.1 MiB (1.2%) fs: vfat dev: /dev/sda2
Swap:
  ID-1: swap-1 type: file size: 3.97 GiB used: 12.3 MiB (0.3%) priority: -2 file: /swapfile
USB:
  Hub-1: 1-0:1 info: full speed or root hub ports: 2 rev: 2.0 speed: 480 Mb/s lanes: 1
    chip-ID: 1d6b:0002 class-ID: 0900
  Hub-2: 1-1:2 info: Intel Integrated Rate Matching Hub ports: 4 rev: 2.0 speed: 480 Mb/s
    lanes: 1 chip-ID: 8087:0024 class-ID: 0900
  Device-1: 1-1.3:3 info: Suyin HD WebCam type: video driver: uvcvideo interfaces: 2 rev: 2.0
    speed: 480 Mb/s lanes: 1 power: 500mA chip-ID: 064e:d251 class-ID: 0e02 serial: <filter>
  Hub-3: 2-0:1 info: full speed or root hub ports: 2 rev: 2.0 speed: 480 Mb/s lanes: 1
    chip-ID: 1d6b:0002 class-ID: 0900
  Hub-4: 2-1:2 info: Intel Integrated Rate Matching Hub ports: 4 rev: 2.0 speed: 480 Mb/s
[... 136 lines cut ...]
       resources: irq:18 memory:c0410000-c041ffff memory:c0420000-c042ffff memory:aff00000-aff007ff
  *-network DISABLED
       description: Wireless interface
       physical id: 9
       bus info: usb@2:1.2
       logical name: wlx98482768db02
       serial: 98:48:27:68:db:02
       capabilities: ethernet physical wireless
       configuration: broadcast=yes driver=rtl8192eu driverversion=v5.11.2.1-18-g8e7df912b.2021052 firmware=0.0 link=no multicast=yes wireless=unassociated
```

**I don't understand why inxi -nn shows state down and sudo lshw -c network shows network DISABLED?**
If I try bringing the interface up it doesn't change what is shown by inxi -nn
```
ed@ed-AO756:~$ sudo ifconfig wlx98482768db02 up
ed@ed-AO756:~$ inxi -nn
Network:
  Device-1: Broadcom BCM4313 802.11bgn Wireless Network Adapter driver: wl
  IF: wlp3s0 state: up mac: e0:06:e6:20:30:c8
  Device-2: Broadcom NetLink BCM57785 Gigabit Ethernet PCIe driver: tg3
  IF: enp4s0f0 state: down mac: b8:88:e3:40:48:e2
  Device-3: TP-Link TL-WN823N v2/v3 [Realtek RTL8192EU] driver: rtl8192eu
    type: USB
  IF: wlx98482768db02 state: down mac: 98:48:27:68:db:02
```

clnhub, 2024-12-30:
It seems that your system already has a wireless driver loaded: wl0. Is it possible to remove this driver with `rmmod`, or `modprobe -r`? Or is is possible to disable the broadcom device in the bios?

Otherwise you probably have to blacklist it.

Could you post output of the `./wifi_install.sh` script and output of the `lspci` command?

enelson1001, 2024-12-31:
Was able to get it working with your suggestions.
Did the following.
1. I reinstalled Linux Mint
2. sudo apt update
3. sudo apt upgrade
4. sudo apt-get install linux-headers-generic build-essential dkms git
5. git clone https://github.com/clnhub/rtl8192eu-linux.git
6. cd rtl8192eu-linux
7. sed -i -E 's/CONFIG_WIFI_MONITOR\s*=\s*n/CONFIG_WIFI_MONITOR = y/' Makefile

8. sudo rmmod 8192eu rtl8xxxu r8188eu wl
9. echo "blacklist rtl8xxxu" >> ./blacklist-rtl8xxxu.conf
10. sudo mv ./blacklist-rtl8xxxu.conf /etc/modprobe.d/
11. echo "blacklist wl" >> ./blacklist-wl.conf
12. sudo mv ./blacklist-wl.conf /etc/modprobe.d/
13 ./install_wifi.sh

14. Reboot

``` 
ed@ed-AO756:~$ iwconfig
lo        no wireless extensions.

enp4s0f0  no wireless extensions.

wlx98482768db02  unassociated  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency=2.412 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
```
**TL-WN823N is still unassociated.**
Click wireless icon on bottom right of screen  (shows to arrows now because wifi is not connected).  Select your wifi network (SSID) and enter password.  The TL-WN823N will now show connect to wifi network.

```
ed@ed-AO756:~$ iwconfig
lo        no wireless extensions.

enp4s0f0  no wireless extensions.

wlx98482768db02  IEEE 802.11bgn  ESSID:"EdAndMary"  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency:2.437 GHz  Access Point: BC:99:11:57:46:CC   
          Bit Rate:144.4 Mb/s   Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
```

Set to Monitor Mode
```
1. sudo iwconfig
2. sudo ifconfig wlx98482768db02 down
3. sudo iwconfig wlx98482768db02 mode monitor
4. sudo ifconfig wlx98482768db02 up
5. iwconfig  -> Should show Mode:Monitor
```

**Now need to figure out why Wireshark does not show eapol frames or http frames even though I have set IEEE802.11 decryption keys**
If you have any suggestions I would appreciate it.

enelson1001, 2025-01-01:
**Was able to successfully monitor my esp32 wifi device with Wireshark**
1. sudo apt-get install wireshark
2. sudo usermod -a -G wireshark YOUR_USER_NAME
3. sudo apt install aircrack-ng
4. iwconfig -> get wifi interface name -> for me was wlx98482768db02
5. Edit NetworkManager.conf 
5a. sudo xed /etc/NetworkManager/NetworkManager.conf
5b. Add [keyfile] at the end so now NetworkManager.conf looks like the following.
```
[main]
plugins=ifupdown,keyfile

[ifupdown]
managed=false

[device]
wifi.scan-rand-mac-address=no

[keyfile]
unmanaged-devices=interface-name:wlx98482768db02
```
5d. Restart NetworkManager -> sudo systemctl restart NetworkManager
6. Enable Monitor Mode
```
1. sudo airmon-ng check kill 
2. sudo ifconfig wlx98482768db02 down
3. sudo iwconfig wlx98482768db02 mode monitor channel 6
4. sudo ifconfig wlx98482768db02 up
5. iwconfig  -> Should show Mode:Monitor
6. Start Wireshark and select the wifi interface wlx98482768db02 
7. Go to Edit/Prefences../Protocols/IEEE 802.11 -> Enable deccryption, Edit decryption keys -> Add wpa-pwd and key
```
7. Disable Monitor Mode
```
1. sudo ifconfig wlx98482768db02 down
2. sudo iwconfig wlx98482768db02 mode managed
3. sudo ifconfig wlx98482768db02 up
4. iwconfig  -> Should show Mode:Managed
5. sudo systemctl restart NetworkManager
6. ```

closed 2025-01-01 by enelson1001

clnhub, 2025-01-02:
Thanks for you detailed feedback and updates.