clnhub/rtl8192eu-linux #113: AP/VLAN support?

JustAnEric, 2025-02-02:
Output from `iw phy1 info` (which is my TL-WN821N v6 adapter):
```
Wiphy phy1
        wiphy index: 1
        max # scan SSIDs: 9
        max scan IEs length: 2304 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Supported Ciphers:
                * WEP40 (00-0f-ac:1)
                * WEP104 (00-0f-ac:5)
                * TKIP (00-0f-ac:2)
                * CCMP-128 (00-0f-ac:4)
                * CMAC (00-0f-ac:6)
        Available Antennas: TX 0 RX 0
        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * P2P-client
                 * P2P-GO
        Band 1:
                Capabilities: 0x1962
                        HT20/HT40
                        Static SM Power Save
                        RX HT20 SGI
                        RX HT40 SGI
                        RX STBC 1-stream
                        Max AMSDU length: 7935 bytes
                        DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: 16 usec (0x07)
                HT Max RX data rate: 300 Mbps
                HT TX/RX MCS rate indexes supported: 0-15
                Bitrates (non-HT):
                        * 1.0 Mbps
                        * 2.0 Mbps
                        * 5.5 Mbps
                        * 11.0 Mbps
                        * 6.0 Mbps
                        * 9.0 Mbps
                        * 12.0 Mbps
                        * 18.0 Mbps
                        * 24.0 Mbps
                        * 36.0 Mbps
                        * 48.0 Mbps
                        * 54.0 Mbps
                Frequencies:
                        * 2412.0 MHz [1] (30.0 dBm)
                        * 2417.0 MHz [2] (30.0 dBm)
                        * 2422.0 MHz [3] (30.0 dBm)
                        * 2427.0 MHz [4] (30.0 dBm)
                        * 2432.0 MHz [5] (30.0 dBm)
                        * 2437.0 MHz [6] (30.0 dBm)
                        * 2442.0 MHz [7] (30.0 dBm)
                        * 2447.0 MHz [8] (30.0 dBm)
                        * 2452.0 MHz [9] (30.0 dBm)
                        * 2457.0 MHz [10] (30.0 dBm)
                        * 2462.0 MHz [11] (30.0 dBm)
                        * 2467.0 MHz [12] (disabled)
                        * 2472.0 MHz [13] (disabled)
                        * 2484.0 MHz [14] (disabled)
        Supported commands:
                 * new_interface
                 * set_interface
                 * new_key
                 * start_ap
                 * new_station
                 * set_bss
                 * join_ibss
                 * set_pmksa
                 * del_pmksa
                 * flush_pmksa
                 * remain_on_channel
                 * frame
                 * set_channel
                 * connect
                 * disconnect
        WoWLAN support:
                 * wake up on anything (device continues operating normally)
        software interface modes (can always be added):
        interface combinations are not supported
        Device supports SAE with AUTHENTICATE command
        Device supports scan flush.
        max # scan plans: 1
        max scan plan interval: -1
        max scan plan iterations: 0
        Supported TX frame types:
                 * IBSS: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
                 * managed: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
                 * AP: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
                 * AP/VLAN: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
                 * P2P-client: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
                 * P2P-GO: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
        Supported RX frame types:
                 * IBSS: 0xd0
                 * managed: 0x40 0xb0 0xd0
                 * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
                 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
                 * P2P-client: 0x40 0xd0
                 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
        Supported extended features:
```

I was wondering how I can get the AP/VLAN functionality working, because I'm using `hostapd` as my network broadcaster. I have tried the commands below in a consecutive fashion, but to no avail I couldn't get hostapd started:
(wlx2887ba8059b5 is my interface used for broadcasting)
```bash
sudo ip link set wlx2887ba8059b5 up

sudo ip link add dev vlan10 link wlx2887ba8059b5 type vlan id 10

sudo ip link add dev vlan20 link wlx2887ba8059b5 type vlan id 20

sudo ip link set vlan10 up

sudo ip link set vlan20 up

sudo systemctl restart hostapd
# error occurs here, verbose output of "sudo hostapd -dddd /etc/hostapd/hostapd.conf"
```

`sudo hostapd -dddd /etc/hostapd/hostapd.conf`
```
random: getrandom() support available
Configuration file: /etc/hostapd/hostapd.conf
nl80211: Using driver-based roaming
nl80211: Supported cipher 00-0f-ac:1
nl80211: Supported cipher 00-0f-ac:5
nl80211: Supported cipher 00-0f-ac:2
nl80211: Supported cipher 00-0f-ac:4
nl80211: Supported cipher 00-0f-ac:6
nl80211: Using driver-based off-channel TX
nl80211: Supported vendor command: vendor_id=0x1018 subcmd=1
nl80211: Supported cipher 00-0f-ac:1
nl80211: Supported cipher 00-0f-ac:5
nl80211: Supported cipher 00-0f-ac:2
nl80211: Supported cipher 00-0f-ac:4
nl80211: Supported cipher 00-0f-ac:6
nl80211: Supported vendor command: vendor_id=0x1a11 subcmd=4106
nl80211: Supported vendor command: vendor_id=0x1a11 subcmd=4107
nl80211: Use separate P2P group interface (driver advertised support)
nl80211: Enable multi-channel concurrent (driver advertised support)
nl80211: use P2P_DEVICE support
nl80211: key_mgmt=0xd0f enc=0x10f auth=0x7 flags=0x20000003a04facc rrm_flags=0x0 probe_resp_offloads=0x0 max_stations=0 max_remain_on_chan=5000 max_scan_ssids=9
nl80211: interface vlan20 in phy phy1
nl80211: Set mode ifindex 42 iftype 3 (AP)
nl80211: Failed to set interface 42 to mode 3: -19 (No such device)
nl80211: Could not configure driver mode
nl80211: deinit ifname=vlan20 disabled_11b_rates=0
nl80211: Remove monitor interface: refcount=0
netlink: Operstate: ifindex=42 linkmode=0 (kernel-control), operstate=6 (IF_OPER_UP)
nl80211 driver initialization failed.
hostapd_interface_deinit_free(0xaaab01644990)
hostapd_interface_deinit_free: num_bss=1 conf->num_bss=1
hostapd_interface_deinit(0xaaab01644990)
br0: interface state UNINITIALIZED->DISABLED
hostapd_bss_deinit: deinit bss vlan20
br0: AP-DISABLED 
hostapd_cleanup(hapd=0xaaab01646160 (br0))
br0: CTRL-EVENT-TERMINATING 
hostapd_free_hapd_data: Interface vlan20 wasn't started
hostapd_interface_deinit_free: driver=(nil) drv_priv=(nil) -> hapd_deinit
hostapd_interface_free(0xaaab01644990)
hostapd_interface_free: free hapd 0xaaab01646160
hostapd_cleanup_iface(0xaaab01644990)
hostapd_cleanup_iface_partial(0xaaab01644990)
hostapd_cleanup_iface: free iface=0xaaab01644990
```

I posted this here because I didn't know whether it was an issue with the driver or something else.

clnhub, 2025-02-02:
The chipset does not support combining interface modes. Which is why AP/VLAN (Supported interface modes) is  missing from the `iw list` output.

JustAnEric, 2025-02-03:
It isn't missing; 

```
        Supported TX frame types:
                 * IBSS: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
                 * managed: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
                 * AP: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
                 * AP/VLAN: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
                 * P2P-client: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
                 * P2P-GO: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
        Supported RX frame types:
                 * IBSS: 0xd0
                 * managed: 0x40 0xb0 0xd0
                 * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
                 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
                 * P2P-client: 0x40 0xd0
                 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
        Supported extended features:
```

clnhub, 2025-02-09:
The driver must mention AP/VLAN as (software)interface for it to work, otherwise it cannot create a VLAN on the same interface.

closed 2025-12-16 by clnhub