clnhub/rtl8192eu-linux #115: Driver not working with OpenSuse Tumbleweed 20250423 Kernel version 6.14.3-1-default

daproghub, 2025-04-25:
Hi I have tried to install Wifi Driver in OpenSuse Tumbleweed 20250423 Kernel version 6.14.3-1-default. But the driver doesn't work correctly.

when I launch **sudo lshw -c network** I get:

network:0
       description: Wireless interface
       product: 802.11n NIC
       physical id: 9
       bus info: usb@1:2
       logical name: wlp0s20f0u2
       serial: e2:a0:59:a4:40:ac
       capabilities: ethernet physical wireless
       configuration: broadcast=yes driver=rtl8192eu driverversion=v5.11.2.1-18-g8e7df912b.2021052 firmware=35.7 link=no multicast=yes wireless=**unassociated**

Here the **lsusb** output:

 Bus 001 Device 007: ID 2357:0109 TP-Link TL-WN823N v2/v3 [Realtek RTL8192EU]

Here the **lsmod | grep 8192eu** output:

8192eu               2084864  0
cfg80211             1409024  3 8192eu,rt2x00lib,mac80211
usbcore               454656  6 8192eu,xhci_hcd,rt73usb,usbhid,xhci_pci,rt2x00usb

Here the **sudo dmesg | grep -i 8192eu** output:

[    9.383112] [    T665] 8192eu: loading out-of-tree module taints kernel.
[    9.592814] [    T665] usbcore: registered new interface driver rtl8192eu
[    9.596560] [    T672] rtl8192eu 1-2:1.0 wlp0s20f0u2: renamed from wlan0
[  103.846388] [     T75] rtl8192eu 1-2:1.0: Runtime PM usage count underflow!
[  287.165849] [   T2977] rtl8192eu 1-2:1.0 wlp0s20f0u2: renamed from wlan0
[  352.679525] [     T75] rtl8192eu 1-2:1.0: Runtime PM usage count underflow!
[  585.589507] [   T4501] rtl8192eu 1-2:1.0 wlp0s20f0u2: renamed from wlan0
[ 2629.189354] [     T75] rtl8192eu 1-2:1.0: Runtime PM usage count underflow!
[ 2633.290441] [  T36789] rtl8192eu 1-2:1.0 wlp0s20f0u2: renamed from wlan0

Hi think that the rt73usb is  another old Cisco USB Wireless G Adtapter that I used temporarily to connect to the internet.

It seems strange because with another distribution like kdeneon with 6.11 kernel it work fine.

and with  lshw -c network command, instead of wireless=unassociated, I got IEEE 802.11 b/g/n or something like that.

Thanks!

clnhub, 2025-04-26:
Yeah, the other driver (rt73usb, rt2x00lib,  rt2x00usb) is probably interfering.

Are these libraries part of OpenSuse, or did you add them by compiling manually?

You can try without the libraries using `rmmod`, or `modprobe -r`. If that fixes your problem, you can blacklist them.

daproghub, 2025-04-27:
> Yeah, the other driver (rt73usb, rt2x00lib, rt2x00usb) is probably interfering.
> 
> Are these libraries part of OpenSuse, or did you add them by compiling manually?
> 
> You can try without the libraries using `rmmod`, or `modprobe -r`. If that fixes your problem, you can blacklist them.

Thanks for the tip!

Indeed these two drivers **rt2x00lib**, **rt2x00usb** and rt73usb were interfering.

I found that when you install OpenSUSE with the WiFi dongle connected, the OS also installs rt2x00lib and rt2x00usb.

The best way is to install the OS **before** connecting the dongle, and I think it is easier to install the rtl892 drivers and register the MOK key before connecting the WiFi dongle.

Thanks a lot! I hope this can help others too!

Here the correct lsmod | grep 8192eu output I have:

8192eu               2084864  0
cfg80211             1409024  1 8192eu
usbcore               454656  4 8192eu,xhci_hcd,usbhid,xhci_pci

closed 2025-12-16 by clnhub