clnhub/rtl8192eu-linux #116: Not loading a freshly installed Driver from COPR

onyeibo, 2025-08-30:
Please react to the issue raised at this [Fedora Discussion](https://discussion.fedoraproject.org/t/sunwire-dkms-rtl8192eu/34799).

`dkms status` confirms that the driver installed
However, the module is not loading. The dongle is not being detected

sunwire, 2025-08-30:
> Fedora release 44 (Rawhide)
Ah, it's rawhide, so maybe there's something broken in the fedora itself. I try to test it.

> I am guessing the installation scripts need to download the gpg keys for creating the mok.pub?
No.

sunwire, 2025-08-30:
> I try to test it.

Everything works on Rawhide (online) and the private key and cert are present 

```
[root@rawhide dkms]# LANG=C ls -l /var/lib/dkms/
total 12
drwxr-xr-x. 1 root root  176 Aug 30 15:53 8192eu
-rw-------. 1 root root 1704 Aug 30 15:53 mok.key
-rw-r--r--. 1 root root  832 Aug 30 15:53 mok.pub
-rw-r--r--. 1 root root  185 Aug 30 15:53 post_transaction.log
```

```
[root@rawhide ~]# modinfo 8192eu 
filename:       /lib/modules/6.17.0-0.rc3.250826gfab1beda7597.32.fc44.x86_64/extra/8192eu.ko.xz
version:        v5.11.2.1-18-g8e7df912b.20210527_COEX20171113-0047
author:         Realtek Semiconductor Corp.
description:    Realtek Wireless Lan Driver
license:        GPL
rhelversion:    10.99
srcversion:     E9BFDF7F9BD3C746E34177E
alias:          usb:v2C4Ep0104d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2C4Ep0100d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p0126d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p0109d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p0108d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p0107d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2019pAB33d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p3319d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p3312d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp818Cd*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v0BDAp818Bd*dc*dsc*dp*icFFiscFFipFFin*
depends:        cfg80211
name:           8192eu
retpoline:      Y
vermagic:       6.17.0-0.rc3.250826gfab1beda7597.32.fc44.x86_64 SMP preempt mod_unload 
sig_id:         PKCS#7
signer:         DKMS module signing key
sig_key:        23:CF:C6:E9:00:12:47:3B:C7:6E:47:61:CF:BE:34:CB:3F:24:73:1D
sig_hashalgo:   sha512
```

onyeibo, 2025-08-30:
> Everything works on Rawhide (online) and the private key and cert are present
>
> ```
> [root@rawhide dkms]# LANG=C ls -l /var/lib/dkms/
> total 12
> drwxr-xr-x. 1 root root  176 Aug 30 15:53 8192eu [...]

I dont have the mok files on my path. Why is that?

> sig_id:         PKCS#7
> signer:         DKMS module signing key
> sig_key:        23:CF:C6:E9:00:12:47:3B:C7:6E:47:61:CF:BE:34:CB:3F:24:73:1D
> sig_hashalgo:   sha512

I get a series of `parm:` entries after` vermagic:` instead

onyeibo, 2025-08-30:
I am going to uninstall the driver and reinstall from COPR directly (not a cached package)

sunwire, 2025-08-31:
> Which package is responsible for providing the mok files? `dkms`?
Yes. [Private key and certificate are auto generated the first time DKMS is run and placed in /var/lib/dkms.](https://github.com/dell/dkms?tab=readme-ov-file#module-signing)

onyeibo, 2025-08-31:
> PS. Check if you have the openssl package installed.

`dkms-rtl8192eu` package pulls in `openssl`

I removed `dkms` (which also removes all dependencies including `dkms-rtl8192eu`).  I then deleted` /var/lib/dkms `and ran the installation all over again hoping that `dkms` will provide the `mok` files. 

The result is the same -- no `mok` files.  This confirms an anomaly.  The `dkms` package is not providing the `mok` files. Filing a bug against it pronto.

onyeibo, 2025-08-31:
Following the instructions at https://github.com/dell/dkms?tab=readme-ov-file#module-signing

I was able to generate the mok keys, import and enrol them at boot time.

`# mokutil --list-enrolled | grep DKMS`
the above confirms successful enrolment.

However, ....
```
# modinfo dkms_test | grep ^signer
modinfo: ERROR: Module dkms_test not found
```

To cap it up, the Wireless Device still does _NOT_ show up.

sunwire, 2025-08-31:
> How do I rebuild the module?

The easiest way is to remove and install dkms-rtl8192eu rpm package.
Or run dkms command e.g.

```
dkms unbuild -m 8192eu -v 5.11.2.1-5.git20250813.fc44
dkms build -m 8192eu -v 5.11.2.1-5.git20250813.fc44
dkms install -m 8192eu -v 5.11.2.1-5.git20250813.fc44
```
where -m is dkms module name and -v is version